ct-Open-Source/tuya-convert #248: vtrust-flash keeps disconnecting other device

prankousky, 2019-08-07:
Hi,

I was originally going to post a different issue (AP only appearing for a second, then disappearing, no connection possible), but I found a solution for that here in the issues section (to kill `wpa_supplicant` manually). So I did that and the `vtrust-flash` AP seemed stable, as in it let me connect to it. I got an ip assigned on my android phone. But it keeps to constantly disconnect, re-connect, disconnect again in a period of about 1-2 seconds each (connect, 1-2secs, disconnect, 1-2secs, re-connect).

Log output below. I was trying this on a `raspberry 3B+`, as you were. This pi is brand-new; in fact, I ordered it just for `tuya-convert` (though it's always good having yet another pi in the house...). As I just received the pi today, it's system was also fresh.

I changed the `config.txt` according to my device names (ethernet device was called something other than `eth0`), besides that, I followed the instructions from your `README.md` - and manually killed `wpa_supplicant` as described here in the issues (the AP wouldn't stay open if I did not do this).

Would somebody please be so kind to help me with this? I used to always solder pins to flash tasmota, but this device (water/gas valve) I am hoping to use soon makes it difficult to do this and I'd prefer this OTA version anyway :)

Thanks in advance for any help solving this problem.

`smarthack-smartconfig.log`
```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
......^CTraceback (most recent call last):
  File "./smartconfig/main.py", line 23, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/home/pi/gits/tuya-convert/scripts/smartconfig/smartconfig.py", line 61, in smartconfig
    sock.send_broadcast(broadcast_body)
  File "/home/pi/gits/tuya-convert/scripts/smartconfig/smartconfig.py", line 35, in send_broadcast
    sleep(self._gap)
KeyboardInterrupt
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.
```
`smarthack-web.log`
```
Listening on port 80
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.7/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.7/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
Listening on port 80
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
[... 421 lines cut ...]
RTNETLINK answers: File exists
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr <meine_mac_adresse> and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA <meine_mac_adresse> IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <meine_mac_adresse>
wlan0: STA <meine_mac_adresse> RADIUS: starting accounting session B9CBF599789C5A3B
wlan0: STA <meine_mac_adresse> WPA: pairwise key handshake completed (RSN)
wlan0: STA <meine_mac_adresse> IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <meine_mac_adresse>
wlan0: STA <meine_mac_adresse> RADIUS: starting accounting session 15441F4328FB2BAA
wlan0: STA <meine_mac_adresse> WPA: pairwise key handshake completed (RSN)
wlan0: STA <meine_mac_adresse> IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED <meine_mac_adresse>
wlan0: STA <meine_mac_adresse> IEEE 802.11: disassociated
wlan0: STA <meine_mac_adresse> IEEE 802.11: associated
wlan0: AP-STA-CONNECTED <meine_mac_adresse>
wlan0: STA <meine_mac_adresse> RADIUS: starting accounting session A09B5E8CAFA106C8
wlan0: STA <meine_mac_adresse> WPA: pairwise key handshake completed (RSN)
wlan0: STA <meine_mac_adresse> IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED <meine_mac_adresse>
```

phormix, 2019-08-08:
Are you using a Pi3 for this? I had similar issues and it's due to a bug between hostapd and the Pi firmware.

I made a post about it here to try and head off the issue:
https://github.com/ct-Open-Source/tuya-convert/issues/241

I was able to fix it by following the steps here:
https://github.com/raspberrypi/firmware/issues/1117

For me, it works with:

Updating the pi via rpi-update:
  rpi-update 82f49d8dfd9c000a203fb1deae64a3eb87abf3c1

Installing a different hostapd version:
Try grabbing this one
  https://archive.raspberrypi.org/debian/pool/main/w/wpa/hostapd_2.6-21~bpo9~rpt1_armhf.deb

After that, reboot and try again.

prankousky, 2019-08-08:
Thank you @phormix 

I am using a brand new Raspberry Pi 3B+ for this. I have followed your instructions, though it still does not work. My Pi is running headlessly, but I am using ethernet for ssh and its internal wifi chip for tuya-convert.

Are you running this as root or not? Instructions say nothing about `sudo`, but I have tried both and the last time - before following your instructions when I was not able to connect to the AP - it only worked when running as root.

Also, are you using Stretch or Buster? I just realized that I am running Buster, but they only tested it on Stretch. I might set up Stretch Light later, but if this is completely irrelevant (= if you are using Buster and it works) I guess that wouldn't do any good, either.

veeceeoh, 2019-08-09:
Same issue with last build of Stretch on a RPi 3B+.

Tried rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b based on cross link from https://github.com/raspberrypi/firmware/issues/1117#issuecomment-468239436

After reboot I found and killed 2 instances of wpa_supplicant.

With WPA_supplicant disabled, the vtrust-flash AP remained enabled, but I had to wait for the handshake / connection to vtrust-flash with my WiFi device to complete before pressing [enter] to start the pairing procedure.

So that particular rpi-update did the trick, not sure why.

prankousky, 2019-08-11:
Thank you. Current setup

* RPi 3B+
* firmware a08ece3d48c3c40bf1b501772af9933249c11c5b
* that hostapd version from your link above

Now **my phone** will connect to that AP successfully. The device I am attempting to flash, however, does not - or if it does, it will not flash it. I held its power button until the blue LED flashed quickly.

So

* `./start_flash.sh`
* `YES`
* connected my android (successfully)
* held device's power button until flashing quickly
* `ENTER`
```
Sarting pairing procedure in screen
RTNETLINK answers: File exists
Waiting f or the upgraded device to appear
.........................................................................................................................................
```
Eventually those dots stop appearing (at first there is another new dot every couple seconds)

The device I am attempting to flash will change from flashing blue LED to constantly lid blue LED. Neither tuya-convert does anything else, nor is there another AP indicating that the device has been flashed with tasmota.

Am I supposed to get any kind of error or status message **if the device is incompatible**? I did not connect it to the ewlink App before (in fact, I never even installed that app on my phone), only attempted to flash it like this. So perhaps it was already incompatible... but if that was the case, can I somehow find out? 

And can I somehow still use it? I wouldn't mind setting it up with their app once; but I assume if I restrict it's internet access (so it cannot send any spy data to their servers) it will not be controllable at all. Due to its build (it is the gas/water valve) I cannot manually flash tasmota.

kueblc, 2019-08-11:
@prankousky Check if there is anything in your web log. If the device hasn't made any requests it's very likely that your device is not compatible with `tuya-convert`. There is no automated detection if this is the case.

Next I would look into `tuyapi` as this could allow you to control the device locally, and then you could block the device from contacting the cloud via your router or `dnsmasq` or similar.

prankousky, 2019-08-15:
Thank you @kueblc . I have already returned the device as I was close to the period where I wouldn't be able to return it any longer and realized that I would not be able to use it even if I successfully flashed it (because it was not compatible to what I'd need it to do after all).

However, I bookmarked `tuyapi` and will attempt this with `Zemismart KS-811 3 Gang Switch` once I get my hands on one of those in case I cannot use `tuya-convert` on them, either :)

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc