ct-Open-Source/tuya-convert #249: Issue With Flashing to Merkury Smart Switch

ClaireSoftware, 2019-08-08:
Hello! i recently bought a Merkury "Smart Wifi Light Switch" from Walmart. the first time I had this switch, I was able to get it to flash with sonoff-tasmota, but upon getting a new unit, the unit seems not to want to enter upgrade mode. i don't think this is a newer model. It connects to the vtrust-flash network, but doesn't seem to request anything from the web or MQTT server. 

https://pastebin.com/PQjF2ib3 <- smarthack-mqtt.log
https://pastebin.com/ZGa8wMYP smarthack-web.log
https://pastebin.com/zsviDxyn smarthack-wifi.log
https://pastebin.com/X8RjCYRX <- smarthack-smartconfig.log

kueblc, 2019-08-08:
If it connected but didn't make any HTTP requests, it's probably running a patched firmware.

kueblc, 2019-08-08:
Can you share the backup for the other switch?

RobinCutshaw, 2019-09-09:
I too just purchased the Merkury color bulbs at Walmart ($19.88 for two).  I tried to flash and noticed that it hung with "."s printed.  Tcpdump shows that the IP is 10.42.42.26, not the expected 10.42.42.42.  Once I changed the ip in the flash script, it got further but the bulb tries to connect to a3.tuyaus.com using https (port 443) instead of port 80 which the web server is listening on.

Interestingly, nmap shows the bulb only listening to TCP port 6668.

I'm guessing that this is the new firmware.

kueblc, 2019-09-09:
Yes, if it made HTTPS requests, it is most definitely running the new firmware.

You should not change the predefined IP address (`10.42.42.42`), as this is how the script determines that the device is running our intermediate firmware. By changing it, you have tricked `tuya-convert` into believing that the device has the intermediate firmware, which it does not, and done nothing to advance the actual flashing.

Port 6668 is open because it is running the stock firmware, which does not run a web server on port 80, only our intermediate firmware does.

In the future, please open a new issue rather than commenting on an unrelated issue (this was about Merkury Smart Switch, not Merkury Color Bulbs). Closing as the original author has not followed up.

closed 2019-09-09 by kueblc