ct-Open-Source/tuya-convert #250: Updating Sonoff-Tasmota to 6.6.0

rlfnb, 2019-08-08:
Sonoff Tasmota released already on 6th of June a new version (6.6.0), would be great to update the version here as well?

meingraham, 2019-08-08:
6.6.0 had some issues. If you are going to replace it in the package, go ahead and grab 6.6.0.4 at http://thehackbox.org/tasmota/020300/sonoff.bin. And yes, sonoff.bin rather than sonoff-basic.bin. We are recommending to those that do get their device to flash with Tuya-Convert to immediately upgrade to sonoff.bin... so it would just be best to have the package do that directly.

kueblc, 2019-09-28:
Feel free to make a PR otherwise I think the current situation is fine. It is intended that you flash another bin after the conversion is complete.

closed 2019-09-28 by kueblc