ct-Open-Source/tuya-convert #251: device-info.txt: Permission denied

daxkamala, 2019-08-11:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

daxkamala, 2019-08-11:
Hi, I love this hack and had flashed a bunch of devices without problems. Then a Pi upgrade blew hostapd and after much pain I got back to an almost working setup. The problem seems trivial, a Permission denied error. I'm guessing it refers to where the device file should be written to the Pi and I've chmodded everything to open all the doors. Can you help please? I'm guessing also that the 404 error may be the same root cause.
[new 2.txt](https://github.com/ct-Open-Source/tuya-convert/files/3490321/new.2.txt)

daxkamala, 2019-08-11:
Sorry, I'm new to this, 
======================================================
Starting pairing procedure in screen
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!

IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   302  100   302    0     0   9741      0 --:--:-- --:--:-- --:--:-- 10066
======================================================
Getting Info from IoT-device
tee: device-info.txt: Permission denied
Hello, world
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
pi@raspberrypi:~/tuya-convert $ curl http://10.42.42.42/flash3
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 2322, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
pi@raspberrypi:~/tuya-convert $

kueblc, 2019-08-11:
Use `sudo` to run as root

daxkamala, 2019-08-11:
Wow! tyhanks, that was quick and fixed the first problem. I thought a sudo -i would be enough, learn every day. Will open a new one for the second problem.

closed 2019-08-11 by daxkamala