ct-Open-Source/tuya-convert #252: 404 errors on curl...

daxkamala, 2019-08-11:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

daxkamala, 2019-08-11:
Have got the device online and ready but every attempt to curl a new .bin results in a 404 error. Files are there in the files subdirectory.
'''Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   302  100   302    0     0  10785      0 --:--:-- --:--:-- --:--:-- 12080
======================================================
Getting Info from IoT-device
Hello, world
======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
root@raspberrypi:~/tuya-convert# sudo curl http://10.42.42.42/flash3
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 2322, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
root@raspberrypi:~/tuya-convert# sudo curl http://10.42.42.42/flash2
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 2322, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
root@raspberrypi:~/tuya-convert# ^C
root@raspberrypi:~/tuya-convert# sudo curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 1569, in _execute
    result = self.prepare()
  File "/usr/lib/python3/dist-packages/tornado/web.py", line 2322, in prepare
    raise HTTPError(self._status_code)
tornado.web.HTTPError: HTTP 404: Not Found
root@raspberrypi:~/tuya-convert#
'''

daxkamala, 2019-08-16:
OK, I'm assuming this is the response from a new Tuya version that we can't flash. I just flashed an older device and got a good response and a successful backup before flashing the new image. The newer device just responds with "Hello, world".

```Good Response from Whitebox Model M6

======================================================
Starting pairing procedure in screen
/home/pi/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
.
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  30206      0  0:00:34  0:00:34 --:--:-- 28936
curl: Saved to filename 'firmware-cd5899.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: cd5899
MAC: 60:01:94:CD:58:99
BootVersion: 4
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

======================================================

Bad Response from new Blitzwolf BW-SHP6
======================================================
Starting pairing procedure in screen
/home/pi/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!

IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   302  100   302    0     0  10066      0 --:--:-- --:--:-- --:--:-- 11185
======================================================
Getting Info from IoT-device
Hello, world
======================================================
```

manju-rn, 2019-08-20:
Can you check if your connected device has the IRC port 6668 open?  If yes, then the Smartconfig has not yet worked its magic and Tuya Firmware is blocking it hence no 80 port is opened.  Unfortunately, need to wait for the guys here to come out with some solution.

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc