ct-Open-Source/tuya-convert #253: Can the tuya.force_update from Home Assistant be used for pushing custom firmware?

manju-rn, 2019-08-17:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_
No
**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_
Yes
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
Yes, BSD29 Smart Plug
**Please remember to close your issue when the problem has been addressed!**
I was unable to get the BSD29 UK Smartplug flashed. It does get connected to the vtrust network but the "Dots" keep on running.  Hence, I configured the Smartlife and got this Smartplug connected to Home Assistant.  I realized that Home Assistant recognizes a Service called "Force_update" from Tuya. I assume that calling this service would update the latest firmware to Smart Plug.  Hence,  I was wondering, if somebody can decode the JSON data to be passed through this "Force_update" service which would update the Smart Plug with custom firmware like Tasmota or ESP Home

EverythingSmartHome, 2019-08-22:
Unfortunately no, according to the docs the "force_update" service is actually for refreshing device information so update in this sense refers to updating account information.

https://www.home-assistant.io/components/tuya/

kueblc, 2019-08-22:
Thanks @EverythingSmartHome for looking into this.

closed 2019-08-22 by kueblc