ct-Open-Source/tuya-convert #254: Flashed successfully, but switch doesn't actually turn on/off anymore

alex-phillips, 2019-08-17:
I've flashed the devices successfully. They are running the sonoff web UI, connected to WiFi just fine. But now if I toggle the switch either via the web UI or even physically on the switch itself, it doesn't actually flip on/off.

meingraham, 2019-08-17:
After you flash the device, you have to configure the firmware for the specific device you have. Check out the Templates article in the Tasmota wiki.

alex-phillips, 2019-08-17:
Got it. I got it to work with the TanTan WP3 plug, but my Gosund WP3 plugs don't seem to be working with the template in the wiki. Any ideas if anyone else encountered this?

kueblc, 2019-08-17:
Closing as this is not a `tuya-convert` issue. Seek advice through Tasmota support channels as @meingraham suggests.

closed 2019-08-17 by kueblc