ct-Open-Source/tuya-convert #255: Hyleton 313 (UK) does not flash - dumped firmware attached

dxnkmrdi, 2019-08-18:
Got some fresh-out-of-the-box Hyleton 313 (UK) plugs.  None of them would flash via OTA.  I suspect these are the "new" firmware versions that are not compatible.  I did not install the provided app so I do not know what version number it claims to be.  These plugs were purchased from Amazon UK in mid August.

I dumped the original firmware before reflashing via soldering.  Hopefully the dump might be some help to get things working again for others - I saw that people were asking for the new firmware but I could not work out what the main tracking issue was so I created this issue the hope that it is useful for others working on this project.

Dump command used:

```
> esptool.py --port COM5 read_flash 0x0 0x100000 hyleton-313-1M.bin
esptool.py v2.7
Serial port COM5
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 
Uploading stub...
Running stub...
Stub running...
1048576 (100 %)
1048576 (100 %)
Read 1048576 bytes at 0x0 in 95.0 seconds (88.3 kbit/s)...
Hard resetting via RTS pin...
```
Thanks to https://znanev.github.io/Hyleton-313-Smart-Plug/ for the hardware guide (although the Sonoff configuration appears to be incorrect now, even if it does install ok).

[hyleton-313-1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/3513539/hyleton-313-1M.zip)

znanev, 2019-08-23:
I can't believe how fast these Tuya SDK based devices are selling! It is practically impossible to buy "old stock", having non-patched version of the firmware. Two months ago I bought some energy measuring sockets from Amazon UK and they too had the patched firmware.. so opening / soldering / serial flashing is the only way to replace the original firmware currently.

Anyway, I'm attaching a dump of the original firmware of Hyleton-313 (I dumped it more than a year ago - file is dated 10 June 2018). If anyone has the skills to decompile these firmware dumps, it might be possible to compare the changes and derive a working OTA flash mechanism eventually.

[hyleton-313-1M-noSSL.zip](https://github.com/ct-Open-Source/tuya-convert/files/3536448/hyleton-313-1M-noSSL.zip)

P.S.: @dxnkmrdi Thanks for the heads up! I'll update the project log with current settings later (templates are the way to go now, here's mine: `{"NAME":"Hyleton-313","GPIO":[57,255,56,255,255,255,255,255,255,17,255,21,255],"FLAG":1,"BASE":18}` ).

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc