ct-Open-Source/tuya-convert #259: Kohree Smart Light Bulb - Can't Flash

speedyrazor, 2019-08-26:
Hi, I have an Kohree Smart Light Bulb ([https://www.amazon.co.uk/gp/product/B07H258JCL/ref=ppx_yo_dt_b_asin_title_o08_s00?ie=UTF8&psc=1]) which I am trying to flash via OTT method to Tasmota, as I have done with many other devices, and trying to use tuya-convert, I have pu the device in pairing mode (fast flashing), but then the tuya-convert doesn't seem to pick up the device. Any help would be much appreciated.

mkormendy, 2019-09-01:
HA! I spent the past weekend attempting to flash this bulb as well!

I went and updated the wiki yesterday after my results and updated it:
https://github.com/arendst/Sonoff-Tasmota/wiki/Kohree-VHP560-Smart-Light-Compatible,-RGBCW-Wi-Fi-LED-Bulb-A19

I've opened up the bulb, with a lot of force I managed to lift the PCB with the LEDs to expose the MCU and decided that the four bulbs I had weren't worth all of the effort to figure out and physically flash the bulb with a UART. I wanted to put in about 7 more bulbs so that would have been 11-12 bulbs and not worth my time. Now seeking another brand of bulb that can be flashed OTA.

For the time being, I am using homebridge in my specific setup with the homebridge-tuya-lan process. I had some issues at first with attempting to get the ID and key for the bulb with the Smart Life app and instead downloaded the CE Smart app and was able to follow the steps to get the ID and keys for the bulbs (this still uses the cloud though 👎  ).
More info here:
https://github.com/AMoo-Miki/homebridge-tuya-lan/issues/63

@kueblc, can you create a editable wiki page of devices that are still working that the public can update, with a simple template that has:

- device name and model
- device version
- device description and notes
- purchasable link(s)

kueblc, 2019-09-28:
Happy to share that support for HTTPS firmware is in the works #279

closed 2019-09-28 by kueblc

mkormendy, 2019-09-29:
@kueblc I'm going to attempt a tuya-convert with the new https firmware support. I'll circle back re: results.

mkormendy, 2019-09-30:
@speedyrazor @kueblc, I got this working (https://github.com/ct-Open-Source/tuya-convert/pull/279#issuecomment-536389958) with the new-api branch and the small code modification to remove the `* 10` from line 140 in the `fake-registration-server.py` file.

Thanks for your help @kueblc!