ct-Open-Source/tuya-convert #260: Bricked SP23

wiredworm, 2019-08-28:
Hi,

Not an issue as such but more of a query.

I tried flashing a Teckin SP23 I had lying around at home this evening and due to a lapse in concentration I missed the step where I should flash the loader to user2.

I went straight to the step where I tried flashing flash3 but it didn't seem to proceed (likely because the loader wasn't in the correct mode).

When I realised the mistake and tried to flash the loader I got an error because the device was no longer replying on the IP it had grabbed.  It now looks like I have a very dead Teckin plug.  When I plug it in and power it on the blue light on the side flashes once and then goes out, but further presses of the button seem to have no effect.  The attached device isn't powered on or off and holding it in won't cause the device to go into pairing mode.

Have I now got a nice new paperweight or is there any steps that could be taken to salvage the unit?

Grateful for any feedback.  It'll teach me to be more awake when flashing firmware onto devices.

Pete

kueblc, 2019-09-04:
`/flash3` will not proceed if you need to do `/flash2` first, so don't worry. You might already have Tasmota installed. Try seeing if there's a new AP `sonoff-****` when you have it plugged in.

wiredworm, 2019-09-05:
> `/flash3` will not proceed if you need to do `/flash2` first, so don't worry. You might already have Tasmota installed. Try seeing if there's a new AP `sonoff-****` when you have it plugged in.

Thanks.  Sure enough I checked and it was there.  I've now got it on the right wifi network.

Only issue now is that clicking toggle in the Web UI has no effect.  There is no relay clicking sound.  I've changed the template to Teckin but no luck.  Hmmmm.

wiredworm, 2019-09-05:
Never mind.  All sorted :-)

The installed Tasmota version wasn't the full featured version so I did an OTA update to bring it up to the 6.6.0 version of release-basic.  I also then found the template needs to be BlitzWolf SHP (45).

Updated that and now it's working fine.

kueblc, 2019-09-05:
Excellent, glad to hear it

closed 2019-09-05 by kueblc

wiredworm, 2019-09-05:
> Excellent, glad to hear it

Many thanks for your help :-)