ct-Open-Source/tuya-convert #261: Device Hangs

razzam21, 2019-08-31:
I have been trying to convert some ZooBee2 SA102 plugs.  I have been able to convert 2 of the 4.  I am working on the third one.  I have a seperated putty window where I can monitor the 4 log files in the scripts directory.  Using the command tail -f *.log.  This allows me to watch what is going on while the other screen runs the start_flash.sh script.

When I put in the third plug it I saw that it was going in the logs.  So I sat and waited and waited.  When it took sigificantly longer then the previous I looked carefully at the logs and I saw the smarthack-web.log file end with 
`[I 190831 15:43:41 web:2246] 200 POST /gw.json?a=s.gw.upgrade.updatestatus&gwId=18480742cc50e3cf03eb&t=100&sign=55ef3a782460d2df7b6497fd1778d766 (10.42.42.18) 7.09ms
[I 190831 15:43:45 web:2246] 206 GET /files/upgrade.bin (10.42.42.18) 39.70ms
[I 190831 15:43:46 web:2246] 206 GET /files/upgrade.bin (10.42.42.18) 911.41ms
`

And the smarthack-mqtt.log file end with this
`1567262625: Received PUBLISH from 18480742cc50e3cf03eb (d0, q0, r0, m0, 'smart/device/out/18480742cc50e3cf03eb', ... (80 bytes))
1567262626: Received PUBLISH from 18480742cc50e3cf03eb (d0, q0, r0, m0, 'smart/device/out/18480742cc50e3cf03eb', ... (81 bytes))
1567262671: Client 18480742cc50e3cf03eb has exceeded timeout, disconnecting.
1567262671: Socket error on client 18480742cc50e3cf03eb, disconnecting.
`
I have left things as is because I don't want to brick the device.  I have tried pinging the IP address listed in the logs and the IP address it should go to when successful. 10.42.42.42 and nothing.  So I scanned the network and I don't see any other IP address (except for laptop that is connected to the vtrust-flash network.

Have I bricked my device?  If not what can I do next as it appears from the logs that it was upgrading it?  I am a software engineer / network engineer for a profession and I am willing to get in the weeds to track this down.  This is my first attempt into the waters of tuya-covert and reading the other bugs didn't help me solve this issue I am having.  Thanks in advance.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3562560/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3562561/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3562562/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3562563/smarthack-wifi.log)

kueblc, 2019-09-01:
It certainly does look like the device attempted to install the [intermediate firmware](https://github.com/vtrust-de/esp8266-ota-flash-convert). As long as you did not cut the power during flashing it's unlikely that it would be bricked. 

If it successfully installed, it will first attempt to connect to `vtrust-flash` at the statically defined address `10.42.42.42`, and if that fails it will spawn an AP `vtrust-recovery`.

If it did not install successfully you should most likely still be running the Tuya firmware, which will have TCP 6668 open.

razzam21, 2019-09-01:
Thank you for your help.  I restarted everything and after starting the process and plugging outlet back in and connected...It did hang on the backup but I did that manually and then triggered the flashing and it worked.   Thanks again for your help.

kueblc, 2019-09-01:
Great, glad it worked out :+1:

closed 2019-09-01 by kueblc