ct-Open-Source/tuya-convert #262: Unclear purpose of flash endpoints / upgrade.bin source?

lheckemann, 2019-09-04:
The README isn't very clear about what exactly flash2 and flash3 do. The questions it raises for me are:

- Is there a way to "test" a custom firmware, loading it only into the device's RAM so that it'll be restored when it's unplugged? (I suspect there will be some size constraints if this is possible at all?)
- How can I restore a backup created by downloading 10.42.42.42/backup?
- What's the difference between flash2 and flash3?

I could probably answer these questions myself given the source code for upgrade.bin, but alas, I couldn't find that, leading me to one last question:
- Is the source code of upgrade.bin available somewhere?

I'd be glad to submit a PR improving the README given answers to my questions :)

kueblc, 2019-09-04:
>     * Is there a way to "test" a custom firmware, loading it only into the device's RAM so that it'll be restored when it's unplugged? (I suspect there will be some size constraints if this is possible at all?)

No

>     * How can I restore a backup created by downloading 10.42.42.42/backup?

You can restore it manually via hard wire programming using `esptool`. The backups are also useful for research purposes.
 
>     * What's the difference between flash2 and flash3?

The 1MB flash is split into two partitions, `userspace1` and `userspace2`, wherein one is the active firmware and the other stores the firmware uploaded OTA. You can only install firmware to the inactive partition. Device info will let you know what your `userspace` is.

`/flash2` checks if the intermediate firmware is running in `userspace2` and installs it if not, otherwise does nothing. The `curl` response indicates if it needs to install and reboot.

`/flash3` then installs the desired firmware image to `userspace1`, along with installing the bootloader. `curl` response indicates if you need to run `/flash2` first.

These might be combined at some point. If either `/flash` installs, you cross the point of no return with `/undo`.

>     * Is the source code of upgrade.bin available somewhere?

Dup of #32, the source is at https://github.com/vtrust-de/esp8266-ota-flash-convert

lheckemann, 2019-09-05:
Thank you for you quick and helpful answers!

closed 2019-09-05 by lheckemann