ct-Open-Source/tuya-convert #263: New version of Tecken SP23

spikeeee, 2019-09-05:
Issue: Newer version of Teckin SP23 does not flash with Tuya-Convert

Teckin contacted me to trial some new versions of their SP23 UK version, the plug is raised slightly so it does not interfere with the switch on the socket.  You can see the physical difference below:

![Image of Teckin SP23](https://lh3.googleusercontent.com/o5zAphjymNyDL9DswXoRGSU3d3ztvqWa7SockgE7h8dlY6MyFCqD_4ppdw3PgVIT_DQ4HIJZsgARsjE7wqu9-QU4GP5EX3Tm68tGrDHChmHqCMPbFZW2wvXl_tSSt6vcb2msNAj3HIdphNV13vNqb5OQ0inqDGd2YdHsuBNweuZV_4c6ZoEtRT7XEn-yAU9_TqnMcrdVyFm6gbje3R0B2Yrfw1Ijg63W_5jX9sNN71GpOzz5_kjNCHOzhf0Z7JY6_3MpCTILE1c_Gu7_3fCHkBGxNThvw9CfE4uoVEybTVLCwZrAbrkdGzG2bP4sGrzGiOLuXeP4iEtz7Sx11isV29t9iO55j7VckpKyw_NxCgGFbWyaOK185RkGGjKlWpt0N0fOVBGWtLrjGZJl0rhLMp2omT1Xrp5vvJbx162VKWpvNP-4FC4LqOfz2KUpAxrVVV9vcJJSfbhAtWoFEM3-f6DlCMjDFoxPUerkVdB8OJ3_VvUZjXpbI8dgTrWAtQpuEp3Aj83CtcnUxemu4Bc6rZmbls84giGo54LF9uiIcSu8EFp5WBEpLcJYAlzZBj8bIpxIOdw2AhQ8I2KghgwRW_AkjuEQ0uamo8VaInG7_KYFyMYdMqcoUJf917xDBecwXJDQZdVuaMaEMNTShmvy1j2zEIP5DUmcwdkR3IlHNc8t9uLSELfFgxLm=w651-h867-no)

Connected one of them to their app for testing.  App reports;

WiFi Module: 1.0.1
MCU Module: 1.0.1

Tried to flash using a RPi3 running Buster.  This same unit with same software was used to flash my previous set of SP23 which flashed successfully.

When running the script I get an endless ..... smarthack-wifi.log shows: 

`Using interface wlan0 with hwaddr b8:27:eb:78:01:36 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA ac:37:43:4d:58:6d IEEE 802.11: associated
wlan0: AP-STA-CONNECTED ac:37:43:4d:58:6d
wlan0: STA ac:37:43:4d:58:6d RADIUS: starting accounting session 90B94E9A9B9620AB
wlan0: STA ac:37:43:4d:58:6d WPA: pairwise key handshake completed (RSN)
wlan0: STA bc:dd:c2:43:01:7f IEEE 802.11: associated
wlan0: AP-STA-CONNECTED bc:dd:c2:43:01:7f
wlan0: STA bc:dd:c2:43:01:7f RADIUS: starting accounting session 0390CBD7D60E0B9D
wlan0: STA bc:dd:c2:43:01:7f WPA: pairwise key handshake completed (RSN)
`
Blue LED on the SP23 stops blinking and goes solid.  However nothing further, just endless ........ and nothing else shows in the logs.

Happy to crack one and dump if someone can advise how to dump it (still learning all of this)

danmed, 2019-09-22:
They did the same with me, i cant get these things to flash. I don't even get the light to go solid.. just endless flashing.

I've tried the new PR for the new sdk too but no go.

kueblc, 2019-09-28:
Please try #279 and leave feedback there, thanks and good luck

closed 2019-09-28 by kueblc