ct-Open-Source/tuya-convert #264: Unresponsive flashed smart switch in endless power loop

simondib, 2019-09-06:
Have just tried to flash a Tuya single gang switch and it's gone into a endless power toggle where it flips between on/off state randomly and persistently. 

I've tried killing the mains with no luck. It's barely responsive to any manual toggle and I can't get it back into detection mode.

I can see a `sonoff-6462` on the WiFi network now, as well as a vtrust-recovery. Occassionally four "Smartlife" devices appear on the network, then later disappear.

It's all a bit strange - would appreciate some help, even just the restore the backed up firmware.

Here's the result of `./start_flash.sh`

`
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  15235      0  0:01:08  0:01:08 --:--:-- 28835
curl: Saved to filename 'firmware-21d93e.bin'
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 21d93e
MAC: XXXXXXXXXXXXXXXXX
BootVersion: 4
BootMode: normal
FlashMode: 1M QIO @ 40MHz
FlashChipId: 1440c8
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
`

simondib, 2019-09-06:
Forgot to mention the model is a unbranded SM-SW102-2 switch

(similar to this: https://github.com/arendst/Sonoff-Tasmota/wiki/Xenon-SM-SW102-2-two-relay-light-switch)

kueblc, 2019-09-07:
Your device flashed Tasmota successfully and is now awaiting configuration. Connect to the `sonoff-6462` AP and follow directions from Sonoff-Tasmota. Good luck!

closed 2019-09-07 by kueblc