ct-Open-Source/tuya-convert #265: Vtrust-Recovery

Mitimin94, 2019-09-07:
Hallo,

habe alles so gemacht wie in der Anleitung jetzt hab ich aber folgendes problem:
die Wlan Birne hat sich verbunden mit dem tool hat auch irgendwas runtergeladen scheinbar,
aber jetzt öffnet die birne den wifi ap vtrust-recovery, habe das Problem natürlich auch gegoogelt aber bin leider zu keinem Ergebniss gekommen was mir wirklich weiter geholfen hat, deswegen jetzt auch die die letzte möglichkeit sich an euch zu wenden der ersteller der ganzen Sache

Mit freundlichen Grüßen Alexander Weber

kueblc, 2019-09-07:
Seeing `vtrust-recovery` means that the intermediate firmware flashed successfully, but was unable to connect to `vtrust-flash`. Simply restart the device while `start_flash.sh` is still running and it should continue with the backup, after which you can continue on to flash custom firmware.

kueblc, 2019-09-14:
Let us know if you have any more trouble

closed 2019-09-14 by kueblc