ct-Open-Source/tuya-convert #267: New Digoo NX-SP202

alfem, 2019-09-13:
I tried Tuya-convert with a brand new Digoo NX-SP202 dual socket but it did not connect to fake AP.

After some sniffing, I found out my socket is attemting to connect to "tuya_mdev_test2".

Then I changed the configuration in hostapd to use this new SSID, but wifi key must have been changed as well because the socket does not connect.

kueblc, 2019-09-13:
Could you please post your logs? Also worth trying the `create-ap` fork, by running `git clone -b create-ap https://github.com/kueblc/tuya-convert.git`

Related #205

alfem, 2019-09-14:
Tested with the create-ap fork. Here you have the logs:

smarthack-mqtt.log:
```
1568452374: mosquitto version 1.4.15 (build date Tue, 18 Jun 2019 11:42:22 -0300) starting
1568452374: Config loaded from mosquitto.conf.
1568452374: Opening ipv4 listen socket on port 1883.
^C1568452651: mosquitto version 1.4.15 terminating
```

smarthack-web.log:
```
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
[32m[I 190914 11:13:09 web:2246][m 200 GET / (10.42.42.87) 1.52ms
^CReceived SIGINT, exiting...

```
smarthack-wifi.log:
```
^Cap0: interface state ENABLED->DISABLED
ap0: AP-STA-DISCONNECTED 3c:71:bf:3f:88:a9

Doing cleanup.. done
Config dir: /tmp/create_ap.wlp3s0.conf.VzX5xvVm
PID: 12579
command failed: Operation not supported (-95)
Creating a virtual WiFi interface... ap0 created.
Network Manager found, set ap0 as unmanaged device... DONE
No Internet sharing
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlp3s0.conf.VzX5xvVm/hostapd_ctrl
Configuration file: /tmp/create_ap.wlp3s0.conf.VzX5xvVm/hostapd.conf
Using interface ap0 with hwaddr e4:42:a6:28:20:4f and ssid "vtrust-flash"
ap0: interface state UNINITIALIZED->ENABLED
ap0: AP-ENABLED 
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: authenticated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: associated (aid 1)
ap0: AP-STA-CONNECTED 3c:71:bf:3f:88:a9
ap0: STA 3c:71:bf:3f:88:a9 RADIUS: starting accounting session C5C4B8DF28E8D90C
ap0: STA 3c:71:bf:3f:88:a9 WPA: pairwise key handshake completed (RSN)
ap0: STA 48:2c:a0:bb:bd:6c IEEE 802.11: authenticated
ap0: STA 48:2c:a0:bb:bd:6c IEEE 802.11: associated (aid 2)
ap0: AP-STA-CONNECTED 48:2c:a0:bb:bd:6c
ap0: STA 48:2c:a0:bb:bd:6c RADIUS: starting accounting session D2F7C17CFD9E99AD
ap0: STA 48:2c:a0:bb:bd:6c WPA: pairwise key handshake completed (RSN)
ap0: AP-STA-DISCONNECTED 3c:71:bf:3f:88:a9
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: authenticated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: associated (aid 1)
ap0: AP-STA-CONNECTED 3c:71:bf:3f:88:a9
ap0: STA 3c:71:bf:3f:88:a9 RADIUS: starting accounting session 76D41376AA2C5B1A
ap0: STA 3c:71:bf:3f:88:a9 WPA: pairwise key handshake completed (RSN)
ap0: AP-STA-DISCONNECTED 3c:71:bf:3f:88:a9
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: disassociated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: authenticated
ap0: STA 3c:71:bf:3f:88:a9 IEEE 802.11: associated (aid 1)
ap0: AP-STA-CONNECTED 3c:71:bf:3f:88:a9
ap0: STA 3c:71:bf:3f:88:a9 RADIUS: starting accounting session 9C1B5F14EAB9690B
ap0: STA 3c:71:bf:3f:88:a9 WPA: pairwise key handshake completed (RSN)
^Cap0: interface state ENABLED->DISABLED
ap0: AP-STA-DISCONNECTED 3c:71:bf:3f:88:a9
ap0: AP-STA-DISCONNECTED 48:2c:a0:bb:bd:6c

Doing cleanup.. ap0: AP-DISABLED 
nl80211: deinit ifname=ap0 disabled_11b_rates=0
done
```

kueblc, 2019-09-14:
Is `48:2c:a0:bb:bd:6c` your Digoo device? It appears, unfortunately, that you are probably running the newer patched firmware.

alfem, 2019-09-15:
Yes, it is. I did not know there is a new firmware :-(

kueblc, 2019-09-15:
Yup, this is mentioned in the README. You can try flashing manually or using `tuyapi` to control your device locally on the stock firmware.

Closing as duplicate of #118 #205

closed 2019-09-15 by kueblc