ct-Open-Source/tuya-convert #269: Teckin SP22 - new Wifi module?

sknoche, 2019-09-13:
Hi, 
after receiving a new batch of Teckin SP22 and failing to OTA flashing, I opened one of the devices to flash them wired...

I was expecting to find a TYWE2S and connect DI0 to GND to enter flashing mode, 

Check for reference:
- https://github.com/arendst/Sonoff-Tasmota/wiki/CE-Smart-Home---LA-WF3-Wifi-Plug-(TYWE2S)

Instead I stumbled upon this chip:
[image]
[image]

Anyone out there with an idea?

kueblc, 2019-09-13:
I see `TYWE2S` silk screened off to the side in those images, I assume you saw these too. From the top picture it does look like it could still be the right chip. From left to right I see what looks like flash memory, ESP8266, and a crystal. Have you tried hooking up to it?

sknoche, 2019-09-13:
I did, soldered TX, RX, 3v and GND... 
But I failed to locate IO0 to get into flashing mode.

GeorgeIoak, 2019-09-14:
This document should help you locate IO0, https://fccid.io/2ANDL-TYWE2S/User-Manual/Users-Manual-3596121.pdf

sknoche, 2019-09-15:
Thanks for the reference, but the schematics indicate a position of IO0 on the blank side of the chip (with the SP22 sticker attached).

[image]

I have absolutely no idea, how to proceed 😞

kueblc, 2019-09-15:
Do you have better pictures to share?

GeorgeIoak, 2019-09-15:
You have to solder a wire onto the IO0 test pad. Tx and Rx are not test points but pins on the module which might be a little difficult to access unless you can get the bottom side of the PCB that this is soldered into (the green PCB in your picture)
[image]

Here's how I did it on a Gosund WP3. I was able to access Tx/Rx at a different spot on the board since I could get to the back side of the TYWE2S unless I removed it from the board

sknoche, 2019-09-16:
[image]

This is a closer look at the backside of the "not-really-a-TYWE2S" :p

GeorgeIoak, 2019-09-16:
Well the front side has full access to the ESP chip so you'll have to look at the pinout of the [ESP8285](https://www.espressif.com/sites/default/files/documentation/0a-esp8285_datasheet_en.pdf) which shows GPIO0 on pin15 and work from there

sknoche, 2019-09-16:
I guess I'll not be able to solder these fine smd pinouts, I'll have to give it a pass :(. My soldering irons tip isn't even small enough to just touch a single pin 😂

DonStefano, 2019-09-19:
This might be the new [WR2 WiFi Module](https://docs.tuya.com/docDetail?code=K8uhkbx75kg7y) which does not have an ESP chip.

sknoche, 2019-09-20:
I can confirm it's not an ESP, but it seems to be the [WR3](https://docs.tuya.com/docDetail?code=K8uhkbxc5bqj7) instead of WR2, so I guess there is no way to flash tasmota :/...

So, I guess the newer versions of all Tuya devices will become more or less "unconvertable".

sknoche, 2019-09-20:
Maybe I'll just get a couple of TYWE2S and change the module itself 😮

meingraham, 2019-09-20:
@DonStefano can you double-check whether yours indeed is WR2?

sknoche, 2019-09-20:
On mine the folowing details are (difficult to read though) printed:

W302
I119WR1 or T119WR1d
G113A6

DonStefano, 2019-09-20:
I did not open it because I want to return it in case it does not work with Tasmota.

meingraham, 2019-09-20:
Well, regardless of whether its a WR2 or WR3, it won't work with Tasmota. Only ESP82xx based Tuya modules work with Tasmota.

kueblc, 2019-09-28:
Unfortunately if this is the case there's not much we can do. Thanks for sharing and documenting.

closed 2019-09-28 by kueblc

henfri, 2019-11-26:
See https://github.com/ct-Open-Source/tuya-convert/issues/285#issuecomment-558850571

omidjozani, 2020-04-08:
It's W302 wifi chip from Realtec and can not be flashed by Tasmota

arannothc, 2020-10-19:
I bought 4x SP22 recently, they contain a RTL8710

[image]
[image]

I will come up with some solution to use Tasmota in theese Plugs... (Custom Chip or Re-Writing Tasmota to theese Chips) 

https://github.com/pvvx/RtlDuino

formatBCE, 2020-11-08:
Any updates on this? Decided to flash all my Teckin sockets, and appears they all have new chip 👎

Blackest-Nighthawk, 2022-11-16:
Sorry for re- opening this theme.

I found just one solution for this problem,
I bought some TYWE2S modules from AliExpress and here you can see the result.

[image]
[image]
[image]

For this you have to modify the bottom of the case a little bit.

[image]

and this ist the adaption plan.
I can draw it clear if anyone is interested of this.

[image]

All in all not so beauty, I know but it works.
So you need around 30 - 45 min. for this swap.

techman83, 2022-11-17:
@Blackest-Nighthawk - That looks like a TYWE2L pinout, are both sets of pins common? If so, I've been retro-fitting esp32-c3 chips into a bunch of my devices via these -> https://templates.blakadder.com/DT-ESP-C05.html

Close to drop in, generally all I've needed to do is a bit of a sand of the ends if the slot is a bit tight length wise.

Blackest-Nighthawk, 2022-11-18:
@techman83 - I bought 12 peaces of SP-22. 4 peaces was with original TYWE2S, other 4 peaces was with the RTL8710 but with the pinout of TYWE2S. This was easy, I just put the new TYWE2S module into the device.
The last 4 peaces was with the completely different pin layout of the RTL8710 board. This is what  you can see on my photos.
So this was the easiest way for me.
The ESP-32 is very nice and powerful. I used this often for other projects.
Maybe you can poste the changed pin configuration for tasmaota. This is perhaps interesting for other user.