ct-Open-Source/tuya-convert #270: NEW Wi-Fi Module - Not Flashing

v-rejected, 2019-09-14:
During the flashing the Device registers but after the registration everything stops.
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3612942/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3612943/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3612944/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3612945/smarthack-web.log)
[image]

kueblc, 2019-09-14:
Unfortunately this device is running an unsupported firmware

closed 2019-09-14 by kueblc

v-rejected, 2019-09-15:
Is it possible that this firmware will become supported in the future? What would be required of me that this firmware would become supported?