ct-Open-Source/tuya-convert #272: How to diagnoze why Tuya IR doesn't connect to vtrust-flash (no record for device's MAC in logs, though smartphone/laptop connects)

PavelGorbanj, 2019-09-18:
Hi everyone,

Device: Tuya universal Smart IR Hub remote (https://www.aliexpress.com/item/32951202513.html)
Model: UFO-R1
FW version: dunno

I've configured my Orange PI Zero so vtrust-flash is up and running (dunno if it runs "properly").
Both my laptop and smartphone can connect (tried subsequently and simultaneously) and they get IPs from expected range (10.42.42...). I see respective records in tuya-convert wifi log.
I don't see any errors in web/mqtt logs.

However, when I put Tuya into connect mode (led blinks quickly), I don't see any trace in wifi log of any attemp to connect from tuya's MAC.

Is there any way I can diagnose such behavior?

What I did so far:
* I've checked if hostapd.conf meets https://smartapp.tuya.com/tuyasmart/faq:
> Set encryption method as WPA2-PSK and authorization type as AES, or set both as auto. Wireless mode cannot be 11n only.
> enable router’s broadcasting function. (as per what I've read, it's not "disabled" by default, and client isolation is OFF by default.
* I've put Tuya into AP mode (slow blink of led) - I can see new open WiFi and I can connect to it from smartphone (without running tuya app, as I simply don't have it installed)

I will also try to add line into hostapd.conf to disable wmm as advised in https://forum.openwrt.org/t/devices-wont-connect-to-wi-fi-smart-life-tuya/18362
But if smb has any other suggestions to diagnose, please post.

Thank you!

PavelGorbanj, 2019-09-18:
tried with wwm disabled - no luck, no single trace of wifi connect attemp from Tya, at least as per wifi logs

brownleeb, 2019-09-20:
I'm in pretty much the same spot.  
I am not sure what state my device is in, it could be stuck in Stage 1/Backup, but I cant tell.
When I plug it in, I can see a "Smartlife-xxxx" SSID.  I ran the convert, but it never came back after "Waiting for upgraded device to appear".  I dont even know if the smartconfig even worked.

I added logging to dnsmasq, and my device (Gosund WP3) never connects, I dont see any attempts.  I have connected to the above SSID with a laptop, and I can ping the device (192.168.175.1) but cannot connect to a web server or via ssh.

I need to fire up wireshark to see if my device is trying to connect.  Like I said, slow flash puts up the Smartlife SSID, but fast flash does not try and connect to vtrust-flash.

I've been all through the start-flash script, and I just dont see where the initial wifi connect is made that allows smartconfig to do its part of config to set the correct params for the rest of the process to work.

Am I missing something here?  If it wont connect to the AP (which is running just fine), I cant get the rest of the process to run...

kueblc, 2019-09-30:
@PavelGorbanj @brownleeb you'll need the device in EZ config mode (fast blink) for `tuya-convert`. AP config mode (slow blink, `Smartlife-****` appears) will not work.

PavelGorbanj, 2019-09-30:
@kueblc, "However, when I put Tuya into connect mode (led blinks quickly), I don't see any trace in wifi log of any attemp to connect from tuya's MAC." Not sure you have noticed this statement in the original post.

kueblc, 2019-09-30:
I did, I also saw that you were trying AP mode and I wanted to save you time and trouble by letting you know that method isn't supported by `tuya-convert`

kueblc, 2019-09-30:
You can try running `./smartconfig/main.py` again while `start_flash` is still running. This will repeat the configuration process, some devices do not get it on the first try. Have you been able to test `tuya-convert` with any other Tuya devices?

PavelGorbanj, 2019-09-30:
"I also saw that you were trying AP mode" that was to confirm my device has WiFi module working fine

PavelGorbanj, 2019-09-30:
"some devices do not get it on the first try". If I don't see single trace in WiFi log for Tuya, while seeing expected connection entries for smartphone, then I guess it makes no sense to expect any further steps to be performed. So Tuya don't even try to connect to vtrust. That's where I need help - diagnose why Tuya doesn't try to connect to expected WiFi. But I guess it's just new FW that is not compatible with the script.

PavelGorbanj, 2019-09-30:
"Have you been able to test tuya-convert with any other Tuya devices?" Nope, I have only one IR Tuya gate.

kueblc, 2019-09-30:
> But I guess it's just new FW that is not compatible with the script

This wouldn't prevent the device from making the initial connection, and besides we are working on supporting new firmware with #279 

You can try repeating the `smartconfig` with and without another device connected to `vtrust-flash`. It may take several attempts before it picks up the wifi credentials.

PavelGorbanj, 2019-09-30:
thank you, will do, though I think I've tried it in number of combinations

PavelGorbanj, 2019-09-30:
""Have you been able to test tuya-convert with any other Tuya devices?" Nope, I have only one IR Tuya gate.", also, just fyi, though I don't have any tuya device, I've tried the script with blitzwolf's SHPs 2/6 - same issue. Anyway, I will retry as you've recommended. Also, I'll check #279 . Thanks!

kueblc, 2019-09-30:
BlitzWolf SHP2/6 is a Tuya device....

PavelGorbanj, 2019-10-02:
> You can try running `./smartconfig/main.py` again while `start_flash` is still running. This will repeat the configuration process, some devices do not get it on the first try. Have you been able to test `tuya-convert` with any other Tuya devices?

apparently this advice has helped in an interesting way - running main.py has identified installation issue. I've got an error that libcrypt is not installed. What's interesting is that initial installation logs contained no complains on this matter. 
After I've installed libcrypt (I had to install python-dev, etc) and restarted start_flash - the UFO flashed from the 1st attempt.

I dunno if that solely my issue or the issue with Armbian/OrangePi Zero, but maybe others experienced the same.

closed 2019-10-02 by PavelGorbanj

kueblc, 2019-10-02:
Thanks for reporting back, glad it worked out! If you would like you can submit a PR or you can provide me with the names of the packages you needed to install I can add them to the `install_prereq` script.

PavelGorbanj, 2019-10-02:
> Thanks for reporting back, glad it worked out! If you would like you can submit a PR or you can provide me with the names of the packages you needed to install I can add them to the `install_prereq` script.

apt-get install python-dev
pip install pycrypto

I needed "-dev" as I've got errors that "Python.h" was missing.

I think it's these. Just in case, I also ran "pip install numpy", "pip install setuptools" but I'm not sure those were really necessary.

BTW, with your advice on running main.py while start_flash was running, I was able to flash my Blitzwolf SHP2 and SHP6 that were connected to official Blitzwolf app (still struggling with SHP5 that doesn't want to backup upon connection to vtrust)

PavelGorbanj, 2019-10-02:
I can try to submit PR later, but as it will be my 1st PR, it may take some time to figure out what to update and what are caveats:)

kueblc, 2019-10-02:
Thanks @PavelGorbanj, these dependencies have actually already been added in today's release of #279

PavelGorbanj, 2019-10-02:
thanks again for your help, without it I wouldn't fix it that "cheap", as soldering for flashing via physical link is not that easy! I will test #279 on my shp5

kueblc, 2019-10-02:
#279 has been merged, so you will only need to update from the master branch