ct-Open-Source/tuya-convert #273: Support for the new sdk

M4dmartig4n, 2019-09-18:
Hi all,

I did succeed in flashing a device with the new tuya sdk with tuya-convert.
Changes needed are in my own fork, for reference

https://github.com/M4dmartig4n/tuya-convert

It is not ready for a pull-request, since it drops support for older sdk atm.

Enjoy !

EverythingSmartHome, 2019-09-19:
Hey, are you saying this works for new firmware that was previously not compatible?

M4dmartig4n, 2019-09-19:
Yes.
Abviously, I haven't tested it against all new firmware, so your mileage may vary.

cr0ky, 2019-09-19:
I hope it will work for the latest tuya devices, bought some door sensors and stuff that would be handy to flash like this.

naxxxe, 2019-09-19:
I just tried this on a Nedis WIFIP310FWT that couldn't be flashed using the old method. I can confirm that this worked very well

EverythingSmartHome, 2019-09-19:
Amazing job well done! Care to give a brief overview of how you figured it out? Just to satisfy my curiosity, I'm really interested! Thanks!

Chaotic, 2019-09-19:
I'll be checking this out tonight got 4 gosunds with the new firmware on them I'll report back if they work.

cr0ky, 2019-09-19:
So just use the build from CT? And import the recent files from you?

kueblc, 2019-09-19:
@M4dmartig4n Go ahead and open a PR and we can work together to get this merged. Will require a few small tweaks to be compatible with #226. You can reach me directly at my gmail of the same name.

M4dmartig4n, 2019-09-19:
@kueblc I will be afk for a moment, but will do that.

Apart from my coding style ;) , we need to figure a way to identify devices with previous/new sdk in order to provide them the right answers (for example, mqtt messages, and upgrade answers are quite different).

Maybe there is a parameter in the request, I dunno.

A "hack" would be to check the source ip in fake_registration server, since new sdk requests are proxyfied through psk-frontend and thus come from 127.0.0.1.

M4dmartig4n, 2019-09-19:
@EverythingSmartHome I will probably write a small blog post about it in the near future

kueblc, 2019-09-19:
> @kueblc I will be afk for a moment, but will do that.

Do you have Telegram or Signal?

> Apart from my coding style ;) , we need to figure a way to identify devices with previous/new sdk in order to provide them the right answers (for example, mqtt messages, and upgrade answers are quite different).
> 
> Maybe there is a parameter in the request, I dunno.

Sure, check for `et=1`

> A "hack" would be to check the source ip in fake_registration server, since new sdk requests are proxyfied through psk-frontend and thus come from 127.0.0.1.

This would work fine too, this entire project is a "hack" :stuck_out_tongue:

SirRedZ, 2019-09-19:
Thanks for the efforts. I was unable to install as the pre-req libssl-dev seemed to be missing/absent. I tried a Kali-linux environment, along with my own Proxmox/Debian LXC container script previosuly working well for Tuya 1.0. I attempted conversion with Teckin Sp20 outlet with no success. IU can see it finds the device and talks to it, and I would say it engages stage one - but never actually backs up or proceeds. It does the leave the device crippled/bricked and unable to proceed. (that is ok it is a test unit and has serial connections ready to reflash back to stock). I am able to reflash back to stock  firmware which DOES work, and I am able to add it to the APP and it detects v1.0.3 and updates it, blocking Tuya Convert v1 -- as expected.

I have another AliExpress plug that doesn't have a firmware update for it, and therefor I can't test with.

[-=SrZ=-]

kueblc, 2019-09-19:
@SirRedZ can you elaborate on what you mean by "bricked"? You can't put it back into EZ config mode after attempting the process?

Janghou, 2019-09-19:
Another missing prerequisite is `git` ;)

SirRedZ, 2019-09-19:
> 
> 
> @SirRedZ can you elaborate on what you mean by "bricked"? You can't put it back into EZ config mode after attempting the process?

Correct. The module's behavior is modified, and button access is unresponsive. Unplugging and plugging in flashes the light once, but that is it. I recognize this behavior after the bootloader has been applied??? mid-state? Only serial flashing allows me to recover.

kueblc, 2019-09-19:
Since you already have it connected to serial, it would be great to get a backup of the firmware in this state. Did you back it up before reflashing?

ThaStealth, 2019-09-19:
Along with the missing libssl-dev pre-req I noticed another issue with my Smart Connect bulbs.

After connecting it is posting to the following URL a couple of times (and failing):

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=14&uuid=<some UUID>&v=4.1&sign=<some hex string>

It seems to be asking for some sort of configuration, but I'm not sure which

haubke, 2019-09-19:
i'm trying for an hour.  i got the smart bulb in config mode. tuya convert is searching. but no luck. do i need to start pairing this device thrue my phone?if yes wich app do you use?

ThaStealth, 2019-09-19:
@haubke,
- is the lamp flashing quickly (2 times per sec) when you start?
- does it stop flashing after you start the process?
- what does the logs tell

You only need to connect to the WiFi with your phone, no apps or anything else

If it doesn’t switch to continuous and keeps on flashing, repeat the process. It took me 10 times for it to finally to connect

haubke, 2019-09-19:
yes flashing 2 times a sec.
no it wont stop flashing. i will check logs when i tried a few times. i will post logs in a minute.

pjv, 2019-09-19:
`sudo apt-get install libssl-dev`

...and then boom, it worked. Saved me from having to try to solder pins that are about 1/8 away from a tall plastic switch soldered to the pcb.

THANK YOU!!!

haubke, 2019-09-19:
@pjv 
i thought i had it installed already. thank you i will try;)

smadhukarHacs, 2019-09-19:
Hi All, 
seeing this in the smarthack-web.log with the new sdk setup

URI:/gw.json?a=tuya.device.upgrade.silent.get&gwId=63120466c44f33863260&t=26&v=4.1&sign=31d615cf31fbc6a63e0031d0685a06c7
Answer s.gw.upgrade.get
b'{"result":{"auto":3,"size":"506587","type":9,"pskUrl":"https://10.42.42.1/files/upgrade.bin","hmac":"7BA1C83AD95ED48CBAB9A3BA731B79DE1CFC61DEDC12CE84BBD480F49833F5F9","version":"9.0.0"},"t":10,"e":false,"success":true}'
[I 190919 17:29:55 web:2246] 200 POST /gw.json?a=tuya.device.upgrade.silent.get&gwId=63120466c44f33863260&t=26&v=4.1&sign=31d615cf31fbc6a63e0031d0685a06c7 (10.42.42.29) 24.65ms

URI:/gw.json?a=s.gw.dev.timer.count&gwId=63120466c44f33863260&t=167&sign=28d676260773c01b4dbfba9d8a94d556
WARN: unknown request: s.gw.dev.timer.count (/gw.json?a=s.gw.dev.timer.count&gwId=63120466c44f33863260&t=167&sign=28d676260773c01b4dbfba9d8a94d556)
[I 190919 17:32:32 web:2246] 200 POST /gw.json?a=s.gw.dev.timer.count&gwId=63120466c44f33863260&t=167&sign=28d676260773c01b4dbfba9d8a94d556 (10.42.42.29) 5.93ms

The last line keeps repeating

kueblc, 2019-09-19:
This should be resolved when we merge this with #226

SirRedZ, 2019-09-19:
Colin, do you still want a copy of firmware in the 'frozen state' ?

SirRedZ, 2019-09-19:
I thought after reading the comments above I should check my logs.... Maybe it was working and crashed for me?>

URI:/gw.json?a=tuya.device.upgrade.silent.get&gwId=88020554cc50e3cf1954&t=7&v=4.1&sign=fab950f88c43806fa17e9bcca40bdf34
Answer s.gw.upgrade.get
b'{"result":{"auto":3,"size":"506587","type":9,"pskUrl":"https://10.42.42.1/files/upgrade.bin","hmac":"7BA1C83AD95ED48CBAB9A3BA731B79DE1CFC61DEDC12CE84BBD480F$
^[[32m[I 190919 07:53:07 web:2246]^[[m^O 200 POST /gw.json?a=tuya.device.upgrade.silent.get&gwId=88020554cc50e3cf1954&t=7&v=4.1&sign=fab950f88c43806fa17e9bcca$
^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 160, in <module>
    main()
  File "./fake-registration-server.py", line 156, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.5/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 421, in run_forever
    self._run_once()
  File "/usr/lib/python3.5/asyncio/base_events.py", line 1388, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.5/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt

kueblc, 2019-09-20:
@SirRedZ Yes, I can use the backup to hopefully debug what went wrong. Doesn't look like the server crashed, it's a `KeyboardInterrupt` exception, which is what `Ctrl-C` throws (the `^C` at the beginning of the error message).

smadhukarHacs, 2019-09-20:
> This should be resolved when we merge this with #226

Worked with the changes. Thanks guys for all the great work.

SirRedZ, 2019-09-20:
[teckin_sp20_afterfw_convertattempt.zip](https://github.com/ct-Open-Source/tuya-convert/files/3633671/teckin_sp20_afterfw_convertattempt.zip)

Here is the FW in "bricked" state. Here are the exact steps.

1. Module defaulted to original FW (rev =unknown - via serial backup made before touching)
2. Module tested for pairing mode/operation via manual button.
3. Module put in to pairing mode, 6 seconds on power button.
4. Device added to Tuya app per normal use.
5. New FW update v1.0.3 identified, accepted and module upgraded.
6. Confirm normal operation of device.
7. Remove device from APP
8. Attempt Tuya convert 2.0 process.
9. Connect mobile device to vtrust-flash
10. Device in pairing mode as a result of APP removal - GO
..... endless dots, connection made to module lights stop flashing, never flash again.
process aborted, - UNPLUG module
11. Restart Tuya-Convert process, reconnect to Vtrust-flash, start process and then plugin module
F/W backup actually starts - it tries, no data received. Enter a few times, multiple lines showing no data and speed of transfer declining.
12. Restart process, Vtrust-recovery visible, module reports flashed with userspace1 please flash with userspace 2 (after an attempt to flash 3rd party firmware)
13. Flashed Userspace 2- rebooted.
14. Restarted flashing process, unplugged module and plugged back in. Mopdule reports online. Firmware backup fails again (expected - file exists?)

15. JUMP - UP - AND - DOWN - SUCCESS !!!!!

Sonoff-xxx access point visible. IT WORKS. I had to jump through hoops - it works.

This was the device I originally used to report the failure- block also!

GREAT WORK!.

[-=SrZ=-]

SirRedZ, 2019-09-20:
My current conclusion  is brute force (unplug/plugin/restart tuya convert process, etc...) gets it to go eventually.

Lightbulbs are a No-go after trying 4-5 different bulbs Zero response. I did my test module 4 times andDigiBlurDIY did 2 or 3 different devices and all eventually converted. There were all previously confirmed BLOCKED in the past. All were added to the app to make sure they had latest firmware, and then removed from  the app to begin the conversion.

Thanks everyone- I hope this helps!

[-=SrZ=-]

EverythingSmartHome, 2019-09-20:
Unfortunately this didn't work for my light bulbs either, great we are moving forward with this though!

Janghou, 2019-09-20:
Happy to report: I does work for my `Action` LSC Smart bulb.

======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: XXXX
MAC: XXXXX
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

======================================================

ThaStealth, 2019-09-20:
> Happy to report: I does work for my `Action` LSC Smart bulb.
>
> ======================================================
> Getting Info from IoT-device
> VTRUST-FLASH 1.1
> (c) VTRUST GMBH [...]

That is strange, I have the same bulbs and they keep posting to the following URL a couple of times (and failing):

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=14&uuid=some UUID&v=4.1&sign=some hex string

Janghou, 2019-09-20:
I did a tweak to the HOSTAPD settings (./setup_ap.sh):

```
cat <<- EOF >/etc/hostapd/hostapd.conf
	interface=$WLAN
	driver=nl80211
	ssid=$AP
	hw_mode=g
	channel=9
	macaddr_acl=0
	auth_algs=1
	ignore_broadcast_ssid=0
	wpa=2
	wpa_passphrase=$PASS
	wpa_key_mgmt=WPA-PSK
	wpa_pairwise=TKIP
	rsn_pairwise=CCMP
	ieee80211n=1
EOF
ieee80211n=1 adds WIFI N channel support, and channel 9 is better then channel 1 I think.
My impression adding WIFI N is better working then only G
```

Janghou, 2019-09-20:
@ThaStealth, nope.

```
URI:/gw.json?a=s.gw.token.get&et=1&gwId=XX&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":1,"source":"ap","path":"broadcast"}}&t=7&v=3.0&sign=YY

URI:/gw.json?a=s.gw.token.get&et=1&gwId=XX&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":1,"source":"ap","path":"broadcast"}}&t=7&v=3.0&sign=YY
Answer s.gw.token.get
ESC[32m[I 190920 11:20:49 web:2246]ESC[m^O 200 GET /gw.json?a=s.gw.token.get&et=1&gwId=XX&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":1,"source":"ap","path":"broadcast"}}&t=7&v=3.0&sign=YY
8686 (127.0.0.1) 9.44ms

URI:/gw.json?a=s.gw.dev.fk.active&et=1&gwId=XX&other={"token":"00000000"}&t=7&v=3.0&sign=1a87f005b67fe56d2327048aa5a992
92
Answer s.gw.dev.pk.active
READ GW ID 01408602c44f3393189c
ESC[32m[I 190920 11:20:50 web:2246]ESC[m^O 200 POST /gw.json?a=s.gw.dev.fk.active&et=1&gwId=XX&other={"token":"00000000
"}&t=7&v=3.0&sign=YYY (127.0.0.1) 54.88ms

URI:/gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=XX&t=9&v=1.0&sign=YY
Answer tuya.device.dynamic.config.get
READ GW ID XX
ESC[32m[I 190920 11:20:51 web:2246]ESC[m^O 200 POST /gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=XX&t=9&v=1.0&si
gn=YY (127.0.0.1) 5.75ms

URI:/gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=XX&t=0&v=1.0&sign=YY
Answer tuya.device.dynamic.config.ack
READ GW ID XX
TRIGGER UPGRADE IN 10 SECONDS
Trigger upgrade in 10 seconds
ESC[32m[I 190920 11:20:52 web:2246]ESC[m^O 200 POST /gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=XX&t=0&v=1.0&sign=YY (127.0.0.1) 62.27ms

```

Janghou, 2019-09-20:
Sonoff-Tasmota v6.5.0 also flashed successfully
`curl http://10.42.42.42/flash3`

Can now switch the bulb on/off with simple requests: ;)
curl http://192.168.0.xx/cm?cmnd=Power%20off
curl http://192.168.0.xx/cm?cmnd=Power%20on

pandit77, 2019-09-20:
You´re the Män!!

Sucessfully flashed all my SP22 Adapters which all had new tuya FW before!

Thanks a lot!

robtrotter, 2019-09-20:
Worked with two Brilliant Smart (Au) plugs.  One of them needed the fixes in #226 and the other failed until I pulled those changes out.  Really not sure on the difference.

Had some garden lights that seemed to get close.  
The trigger_upgrade.sh ran - but then got a disconnect on the MQTT log:  "1568993822: Socket error on client P1, disconnecting."

SirRedZ, 2019-09-20:
> Worked with two Brilliant Smart (Au) plugs. One of them needed the fixes in #226 and the other failed until I pulled those changes out. Really not sure on the difference.
>
> Had some garden lights [...]

I have experienced that also.

SkipperPit, 2019-09-20:
It works! I just flashed 12 Teckin SP22 smart sockets. THANK YOU!

Janghou, 2019-09-20:
Only Action LSC Smart White bulbs have success, Action Smart Connect RGB bulbs fail to flash here. They only sometimes stop flashing in the initial phase, but then a 404 shows up in ..web.log:
```

ESC[33m[W 190920 18:20:47 web:2246]ESC[m^O 404 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=21&u
uid=yy&v=4.1&sign=xx (127.0.0.1) 60.44ms
ESC[33m[W 190920 18:20:49 web:2246]ESC[m^O 404 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=21&u
uid=yy&v=4.1&sign=xx (127.0.0.1) 54.38ms
```

FYI Smart Connect RGB bulbs have ID's that are 2 characters longer then white filament bulbs, can that be the cause?

JoshuaSG, 2019-09-20:
@haubke I had the same issue you had and was able to fix it by running the following two commands
`sudo apt-get install python3-dev`
`pip3 install pycrypto`

ThaStealth, 2019-09-20:
@janghou

I have the issue, didn’t find a solution yet, however if you google tuya.device.config.get you’ll find a page from tuya itself which looks like an MQTT server info request

kueblc, 2019-09-20:
@Janghou looks like you have a unique firmware there. It would be great to get a copy of it for research.

As far as your current issue, looks like it's making a request to `/d.json` instead of `/gw.json`. Duplicate [this line](https://github.com/M4dmartig4n/tuya-convert/blob/ea815b7d62d164af7bae6bedabc63665b65baa2c/scripts/fake-registration-server.py#L147) and change one of the `gw` to `d`. Let us know how it goes.

ThaStealth, 2019-09-20:
@kueblc 
I have the same bulb, changed the line to test it, but it still doesn't work:

URI:/d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=35&uuid=**removed**&v=4.1&sign=**removed**
WARN: unknown request: tuya.device.config.get (/d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=35&uuid=**removed**&v=4.1&sign=**removed**)
^[[32m[I 190920 19:25:00 web:2246]^[[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=35&uuid=**removed**&v=4.1&sign=**removed**(127.0.0.1) 52.80ms

Note that "tuya.device.config.get" is not implemented in the JSONHandler

kueblc, 2019-09-20:
@ThaStealth thanks for testing and following up. Try removing `.dynamic` from [this line](https://github.com/M4dmartig4n/tuya-convert/blob/ea815b7d62d164af7bae6bedabc63665b65baa2c/scripts/fake-registration-server.py#L77)

Janghou, 2019-09-20:
@ThaStealth  @kueblc

```
URI:/d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=10&uuid=xxx&v=4.1&sign=yy
Answer tuya.device.dynamic.config.get
READ GW ID son?a=tuya.device.co
ESC[32m[I 190920 19:41:18 web:2246]ESC[m^O 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=10&uuid=xx&v=4.1&sign=yy (127.0.0.1) 58.63ms
```

kueblc, 2019-09-20:
@Janghou so, did it work?

digiblur, 2019-09-20:
I've tried about 3 or 4 bulb types with 2.0 and they don't work yet.  One does this..  

RI:/gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=70007862b4e62d5ccf37&t=8&v=1.0&sign=7e4d82fbbf61fb1b0a786abd035778ad
Answer tuya.device.dynamic.config.get
READ GW ID 70007862b4e62d5ccf37
[32m[I 190919 01:10:22 web:2246][m 200 POST /gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=70007862b4e62d5ccf37&t=8&v=1.0&sign=7e4d82fbbf61fb1b0a786abd035778ad (127.0.0.1) 5.47ms

URI:/gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=70007862b4e62d5ccf37&t=0&v=1.0&sign=9849fa1adcceb0159eaf9ec21b309081
Answer tuya.device.dynamic.config.ack
TRIGGER UPGRADE IN 10 SECONDS
Trigger upgrade in 10 seconds
[32m[I 190919 01:10:22 web:2246][m 200 POST /gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=70007862b4e62d5ccf37&t=0&v=1.0&sign=9849fa1adcceb0159eaf9ec21b309081 (127.0.0.1) 60.66ms
b'2.2\x8e\xf6\x14X85542492\xb7\x98\xa7\x08\xc0\xc7\xa5\xc2JB\x9c_9\xed\x91m\x8d\xb2~AK\xa9\xe2\xcf\xe7\x9a\xd8\xd8S\xf25P\xa89\x12\nX\xf1vdR\xd7c\n!\xed\x7fB\x91\xfa\xef\xd4\xa8\xffL\x1a\xef\x01\r\x8e\xbd\xf5EL(\x95\xab\x04\x9c\x1b\xd5\xae=\xaa\xe4\x02\xd5\x0bNl\xcd\xackG\xe0\xf9\x12\x8dT)O\xb0\x9b\x05u\xe6'

ThaStealth, 2019-09-20:
> @Janghou so, did it work?

I tried it and it repeats a couple of times but it doens't continue (got the same output @Janghou has)

Janghou, 2019-09-20:
@kueblc No, hanging on above message. missing gwID in the response I guess, see

**READ GW ID son?a=tuya.device.co**

kueblc, 2019-09-20:
Good to know, thanks for reporting. I'll follow up later tonight.

l3ok, 2019-09-20:
flash fails with the following repeating in the mqtt.log:

1569008459: New connection from 127.0.0.1 on port 1883.
1569008459: Client 08702840840d8ea3877d already connected, closing old connection.
1569008459: Socket error on client 08702840840d8ea3877d, disconnecting.
1569008459: New client connected from 127.0.0.1 as 08702840840d8ea3877d (c1, k60, u'AAAAAI56VcDYRHclIcnZCVaQ').
1569008459: No will message specified.
1569008459: Sending CONNACK to 08702840840d8ea3877d (0, 0)
1569008459: Received SUBSCRIBE from 08702840840d8ea3877d
1569008459:     smart/device/in/08702840840d8ea3877d (QoS 0)
1569008459: 08702840840d8ea3877d 0 smart/device/in/08702840840d8ea3877d
1569008459: Sending SUBACK to 08702840840d8ea3877d
1569008462: New connection from 127.0.0.1 on port 1883.
1569008462: New client connected from 127.0.0.1 as P1 (c1, k60).
1569008462: No will message specified.
1569008462: Sending CONNACK to P1 (0, 0)
1569008462: Received PUBLISH from P1 (d0, q0, r0, m0, 'smart/device/in/08702840840d8ea3877d', ... (111 bytes))
1569008462: Sending PUBLISH to 08702840840d8ea3877d (d0, q0, r0, m0, 'smart/device/in/08702840840d8ea3877d', ... (111 bytes))
1569008462: Socket error on client P1, disconnecting.

URI:/gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=08702840840d8ea3877d&t=8&v=1.0&sign=4737418eb266715d269b52c1d7167bb1
Answer tuya.device.dynamic.config.get
READ GW ID 08702840840d8ea3877d
[I 190920 20:44:57 web:2246] 200 POST /gw.json?a=tuya.device.dynamic.config.get&et=1&gwId=08702840840d8ea3877d&t=8&v=1.0&sign=4737418eb266715d269b52c1d7167bb1 (127.0.0.1) 3.99ms

web log:

URI:/gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=08702840840d8ea3877d&t=0&v=1.0&sign=b666fbf0c00aa28138972a2af1d002d2
Answer tuya.device.dynamic.config.ack
READ GW ID 08702840840d8ea3877d
TRIGGER UPGRADE IN 10 SECONDS
Trigger upgrade in 10 seconds
[I 190920 20:44:57 web:2246] 200 POST /gw.json?a=tuya.device.dynamic.config.ack&et=1&gwId=08702840840d8ea3877d&t=0&v=1.0&sign=b666fbf0c00aa28138972a2af1d002d2 (127.0.0.1) 16.46ms
b"2.2\xc4\xd2\x91\x1900870007\xb7\x98\xa7\x08\xc0\xc7\xa5\xc2JB\x9c_9\xed\x91m\x8c\x1e\x84\x1dY\xde\xae\xcfzh7\xd6\x16)\xe0\x0f\xb3/\x16\xab\xb954v\xd2\xf0\x08\xc7\xf7%\x13\xe1\x91\xfa\xef\xd4\xa8\xffL\x1a\xef\x01\r\x8e\xbd\xf5ELX{\xbc\xaf$\x0f'\xfbz~WV\xb6\xbe\x03>\xfdZ\x08\xc8b\x8f\x06F3\x8c[\xa9'\x99\xae\xe1"

Homer-Sim, 2019-09-20:
Great Job guys!
Where is the moment in tutorial when I have to change something? Big Thanks

jakommo, 2019-09-20:
I think you only need to use the other repo url in the installation when you git clone. 
I.e. in https://github.com/M4dmartig4n/tuya-convert#installation do `git clone https://github.com/M4dmartig4n/tuya-convert.git` instead of `git clone https://github.com/ct-Open-Source/tuya-convert`

SirRedZ, 2019-09-20:
#### 1569008462: Socket error on client P1, disconnecting. 

I am convinced this is my issue. Any ideas for me to try @kueblc ??

Chaotic, 2019-09-20:
Tried this on a raspberry pi 3 tonight and getting this issue when attempting to run the ./install_prereq.sh

`pi@raspberrypi:~/tuya-convert $ ./install_prereq.sh
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://deb.nodesource.com/node_11.x stretch InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version (7.52.1-5+deb9u9).
haveged is already the newest version (1.9.1-5+deb9u1).
mosquitto is already the newest version (1.4.10-3+deb9u4).
net-tools is already the newest version (1.60+git20161116.90da8a0-1).
python3-setuptools is already the newest version (33.1.1-1).
python3-wheel is already the newest version (0.29.0-2).
screen is already the newest version (4.5.0-6).
dnsmasq is already the newest version (2.76-5+rpt1+deb9u1).
hostapd is already the newest version (2:2.6-21~bpo9~rpt1).
python3-pip is already the newest version (9.0.1-2+rpt2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-_t0g0dmb-build
Requirement already satisfied: paho-mqtt in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: pyaes in /usr/local/lib/python3.5/dist-packages
Requirement already satisfied: tornado in /usr/local/lib/python3.5/dist-packages
Installing collected packages: sslpsk
  Running setup.py install for sslpsk ... error
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-_t0g0dmb-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gstdi5kq-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv7l-3.5
    creating build/lib.linux-armv7l-3.5/sslpsk
    copying sslpsk/__init__.py -> build/lib.linux-armv7l-3.5/sslpsk
    copying sslpsk/sslpsk.py -> build/lib.linux-armv7l-3.5/sslpsk
    creating build/lib.linux-armv7l-3.5/sslpsk/test
    copying sslpsk/test/test_sslpsk.py -> build/lib.linux-armv7l-3.5/sslpsk/test
    copying sslpsk/test/__init__.py -> build/lib.linux-armv7l-3.5/sslpsk/test
    copying sslpsk/test/example_client.py -> build/lib.linux-armv7l-3.5/sslpsk/test
    copying sslpsk/test/example_server.py -> build/lib.linux-armv7l-3.5/sslpsk/test
    copying sslpsk/test/__main__.py -> build/lib.linux-armv7l-3.5/sslpsk/test
    running build_ext
    building 'sslpsk._sslpsk' extension
    creating build/temp.linux-armv7l-3.5
    creating build/temp.linux-armv7l-3.5/sslpsk
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c sslpsk/_sslpsk.c -o build/temp.linux-armv7l-3.5/sslpsk/_sslpsk.o
    sslpsk/_sslpsk.c:19:25: fatal error: openssl/ssl.h: No such file or directory
     #include <openssl/ssl.h>
                             ^
    compilation terminated.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-_t0g0dmb-build/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gstdi5kq-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-_t0g0dmb-build/
`

ThaStealth, 2019-09-20:
@chaotic Read this thread, the issue and solution were posted a couple of times already..

MrPaulAR, 2019-09-20:
The fix for the sslpsk failure is to install the libssl libraries.

`sudo apt-get install libssl-dev -y`

MAlexanderRobinson, 2019-09-20:
Thanks, guys... I was able to flash some plugs that were just sitting around that 1.0 failed to flash. Nice work!

[303 more replies]

closed 2019-09-28 by kueblc