ct-Open-Source/tuya-convert #275: Unable to flash ANOOPSYCHE AWP8L wifi plug

BeAr1987, 2019-09-20:
I'm unable to flash a ANOOPSYCHE AWP8L wifi plug bought on amazon recently. I already tried the version from @M4dmartig4n which he claims should work with the new tuya SDK but this also does not succeed.
From the logfiles I can see that the plug connects to the wifi and then disconnects again, but i'm unable to interpret the rest of the log and would appreciate some help, logs attached. 
If it matters, i'm using a RPi Zero W with its onboard wifi chip and the latest Raspbian buster image. 

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3637432/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3637433/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3637434/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3637435/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3637436/smarthack-wifi.log)

SirRedZ, 2019-09-21:
Have you attempted with version 1.0 already?

SirRedZ, 2019-09-21:
If you add the device to the app, does it update the firmware? I recommend doing so if you have not already.

BeAr1987, 2019-09-21:
If you mean if I tried with the original from c't, then yes, this was my first attempt until I saw the comments that tuya brought a new sdk, then I tried with m4admartig4n's fork which also had no success.
The plug wasn't connected to the web or official app yet, but I could try to update. In the last attempt I still can try to get physical access with a FTDI cable, but therefore I would need to break open the plug.

kueblc, 2019-09-28:
Is this resolved?

BeAr1987, 2019-09-28:
Kind of, I hardwired the plug and flashed tasmota directly. I made a backup of the original FW which I wanted to upload here so that maybe someone with more expertise could take a look at it and the plugs maybe could be compatible with tuya convert in the future. 
I currently do not have access to the RPi I've used to pull the backup and that's why I haven't posted this yet.

kueblc, 2019-09-28:
OK thanks for the update! Looking forward to debugging your firmware backup when you get around to uploading it.

BeAr1987, 2019-09-30:
So, attached is the backup. Made  a 1 MB, 2 MB and 4 MB backup all contained in the ZIP file because I didn't knew how big the actual file is/was.
I do have a second plug which is still running on the initial FW because I did not get that cracked open without destroying the housing. So, when you want me to test something, just let me know and I can set someting up. 

[fwbackup.zip](https://github.com/ct-Open-Source/tuya-convert/files/3669627/fwbackup.zip)

kueblc, 2019-11-03:
@BeAr1987 I wasn't able to reproduce any issue with this firmware image, I'm guessing it was most likely an issue with `dnsmasq` since in your logs you can see the device connect to `vtrust-flash` but not make any requests.

If you have any more devices to flash, try updating to the latest and let me know if you experience any more issues. Thanks for the report and the firmware image!

closed 2019-11-03 by kueblc

BeAr1987, 2019-12-27:
@kueblc thanks for looking into this, today I gave it another try and succeeded.
Made a complete new Kali VMware install and it worked with the latest tuya convert software, thanks again!

kueblc, 2019-12-27:
Great to hear, thanks for letting me know!

Testinguy, 2020-02-28:
Hi there,

apologies for the most likely veeery trivial question:
does Tasmota enable power monitoring for all/most smart plugs, or is it a hardware specific capability which plugs such as this Anoopsyche will never have?

I'm looking for a cheap tuya-convert-able EU-plug with reliable power monitoring as main focus... Any advice?

Thanks for your hinsights!
Tom

Testinguy, 2020-02-28:
Hi there,

apologies for the most likely veeery trivial question:
does Tasmota enable power monitoring for all/most smart plugs, or is it a hardware specific capability which plugs like this Anoopsyche will never have?

I'm looking for a cheap tuya-convert-able EU plug with reliable power monitoring as main focus, would you have a pick?

Thanks for your hinsights(/patience!)