ct-Open-Source/tuya-convert #278: Diagnose new sdk?

brownleeb, 2019-09-26:
Working with Gosund WP3 and Kali 4.19.

Had trouble with 'screens' so I just manually run all the commands in different terminal windows.  After running smartconfig, the IoT device connects to the AP, but it gets 10.42.42.23 (instead of the .22 the script expects).  Ping is fine, but curl gets refused connection.  NMAP shows only port open is 6668 for IRC.

I presume this means its the upgraded firmware?

kueblc, 2019-09-26:
> After running smartconfig, the IoT device connects to the AP, but it gets 10.42.42.23 (instead of the .22 the script expects).

This is expected, the device won't take `10.42.42.42` until it's running the intermediate firmware.

> Ping is fine, but curl gets refused connection. NMAP shows only port open is 6668 for IRC.

All this means is that your device is still running stock firmware

> I presume this means its the upgraded firmware?

Not enough information to tell, please do post your logs when opening an issue. We have a template to make it easy to open a new issue with all the requisite information.

brownleeb, 2019-09-27:
Thanks, Colin.  I'll re-run it and see if I can pull the logs.  As I mentioned, I was having trouble with that part of the script, and was working on debugging it myself.
In reality, I was hoping to get a bit more detailed info on the process flow, maybe a wiki or something.  You've done a huge amount of great work, and while I know I am asking for a little more, my intent is to learn enough that I can help.  There was a previous "issue" I read on Arch linux, and I was going to try and help with an Arch port (I run Arch on all my RPis)...
I realize that the smartconfig process is what sets the device to connect to the vtrust-flash SSID.  After that, I presume the device is pre-configured to reach out for updated firmware, and through the dnsmasq, the tuya.com domain is poisoned, and sent to our local server, where the fake_registration_server takes over and drops the intermediate firmware (which I presume has a functioning web server or something that answers on port 80).  
I still havent figured out how the very first step with smartconfig happens.  I see that it uses broadcast and multicast, but I assume somewhere this is some magic to broadcasting packets between SSIDs.  I would guess that the 'typical' install procedure from the manufacturer would use a wifi-direct type connection, initiated from the IoT device side.
I just think a process flow, even just with pointers to wikipedia pages for detailed info (ie, to learn more about wifi-direct) would help people fix their own problems and possibly contribute to the project.  But, like I said, not trying to pile on extra work...

kueblc, 2019-09-27:
You've got almost everything

> I presume the device is pre-configured to reach out for updated firmware

Yes and no, some devices will ask for an update automatically, while others wait for the update signal via MQTT.

> I still havent figured out how the very first step with smartconfig happens. I see that it uses broadcast and multicast, but I assume somewhere this is some magic to broadcasting packets between SSIDs.

The IoT device enters sniffer mode and while the content of the packets are encrypted, the length remains unchanged. So the data is encoded into the lengths of broadcast packets, and as a secondary strategy, also encoded into the IP addresses of multicast packets.

> I would guess that the 'typical' install procedure from the manufacturer would use a wifi-direct type connection, initiated from the IoT device side.

No, in fact the smartconfig procedure is the default. If that fails, the device can spawn an access point which you can configure through, but not wifi direct.

meingraham, 2019-09-27:
Also, this - https://github.com/ct-Open-Source/tuya-convert/pull/277#issuecomment-535912256

brownleeb, 2019-09-27:
Thanks for the background...  Before I ask dumb questions, I like to do my best to resolve...

It turns out my device (as I expected, its new-ish) is trying to connect to port 443 on a3.tuyaus.com (I built a small python server to answer on 443 and print the header).
I am trying the new branch for SSL.  That reminded me that I am out of space on my linux drive, and that I had been meaning to move to a larger stick....

Currently doing the prerequisite install....

brownleeb, 2019-09-28:
Again (for SEO), on gonsund WP3 plug:
Ran the fork [https://github.com/M4dmartig4n/tuya-convert](url) from M4dmartig4n.  Ran great!
Thanks!

closed 2019-09-28 by brownleeb