ct-Open-Source/tuya-convert #280: [wishlist] Helper script for recovering wrongly entered SSIDs

schildbach, 2019-09-27:
I think it would be nice to have a script that helps with wrongly entered SSIDs (or maybe also passwords). The idea is to sniff the WLAN for connection attempts and print them to the console. That way we could find out what a particular device is configured for.

SirRedZ, 2019-09-27:
Although I don't think this is a terrible idea, the latest commit of Tuya convert includes a 'thirdparty.bin' provided by @digiblur which now enables WiFi Manager by default. Successive failures to connect due to incorrect password or SSID will now re-spawn the ' sonoff-xxxx ' AP again! Just make sure you flash ####thirdparty.bin and not another version of Sonoff-Tasmota.

There is also talk that perhaps a WifiMan sonoff bin will be pre-compiled by Tasmota in the future!

[-=SrZ=-]

schildbach, 2019-09-27:
@SirRedZ Hmm, have the security implications of that change been thought through? This means an attacker can disconnect the WLAN router, spawn an own one with the same SSID and then take over all IoT devices in that network.

SirRedZ, 2019-09-27:
Your point is well understood - I guess users would have control to add web admin password to also protect against this. It is so common for people to lock themselves out, in some cases making the module garbage for many users without additional skills and resources. I would say this project is aimed at the greater masses, and as such this is the lesser of the evils. Since the Wifi manager option is reset upon a sonoff updated flash, and web admin password offers another layer of protection, my opinion is the risk is somewhat mitigated. 

Also, both binary files are included in the project, so people have choices to be made!

Interestingly enough, this entire project is about **DEFEATING PROTECTIONS** so it's certainly an oxy-moron indeed! @schildbach 

Have an awesome day!

[-=SrZ=-]

meingraham, 2019-09-27:
In a home automation scenario, hopefully the home owner has physical control of the router thus not allowing it to be powered off or disconnected while the rest of the smart switches in the home remain powered on.

digiblur, 2019-09-27:
@sirredz the wifi manager is not reset after a new bin is written. It's only reset if the user resets to defaults after the new bin is written.

SirRedZ, 2019-09-27:
My mistake - I was thinking that but didn't specify it. I often reset 5, so I kind of forgot to mention that step. Good call-out!

[-=SrZ=-]

kueblc, 2019-09-27:
Nice idea @schildbach, if you submit a PR we can see how it works. Otherwise I believe #277 should cover this. Thanks for sharing your ideas!

closed 2019-09-27 by kueblc

schildbach, 2019-09-29:
Just wanted to document the steps I took to recover from a wrong SSID (using Ubuntu 18.04, on an Intel Wifi):

- sudo apt install aircrack-ng wireshark
- sudo airmon-ng check kill
- sudo airmon-ng start <your wifi device> (e.g. wlp58s0 or wlan0)
- sudo wireshark
- select your wifi device from the list
- plug in the misconfigured device and immediately watch SSIDs, you should see your misconfigured SSID pretty soon