ct-Open-Source/tuya-convert #285: Unable to flash Tekin SP22

floli, 2019-10-03:
Hello,

I try to flash a Tekin SP22 WiFi switch / power meter. I use the `master` branch after #279 has been merged to it, executing on a Raspberry Pi.

Commit is d9211e0f17641b4af46336c69fd067d0742684ec. Raspbian Buster

I run `./start_flash.sh` after a fresh reboot, however, the `vtrust-flash` wifi network does not appear. After stopping and restarting `start_flash.sh` it appears and I can connect to it.

I put the switch into flashing mode by a long press on the button, it starts to blink red-blue.

About two seconds after I press Enter to continue flashing, the switch goes back to normal mode (stops blinking):

```
Starting pairing procedure in screen
RTNETLINK answers: File exists
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
.... dots keep appearing
```
and that's it until I cancel the flashing after a couple of minutes.

The log files:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3685376/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3685377/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3685379/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3685380/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3685381/smarthack-wifi.log)

Thanks your any advice!

kueblc, 2019-10-03:
Looks like it did flash the intermediate firmware. Try unplugging the device and plugging it back in while `start_flash` is still running (the `vtrust-flash` AP is online). Then try to `curl http://10.42.42.42`.

floli, 2019-10-04:
Ok, I made some headway. Don't know exactly why, it seems that my problem was, that on first attempt after the Raspberry reboot the vtrust-flash network was not created. Before my second attempt, I did not execute `stop_flash.sh` before, which somehow made it not working. But now:
```
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  4 1024k    4 49640    0     0  14413      0  0:01:12  0:00:03  0:01:09 14413
curl: Saved to filename 'firmware-71d2e4.bin'
curl: (18) transfer closed with 998936 bytes remaining to read
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 71d2e4
MAC: 84:0D:8E:71:D2:E4
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

======================================================
Please make sure to note the correct SPI flash mode!
Installing an alternative firmware with the wrong flash mode will leave the ESP unable to boot!

Next steps:
1. To go back to the orginal software
   # curl http://10.42.42.42/undo

2. Be sure the conversion software runs in user2
   # curl http://10.42.42.42/flash2

3. Flash a third party firmware to the device
BE SURE THE FIRMWARE FITS THE DEVICE AND USES THE CORRECT FLASH MODE!
MAXIMUM SIZE IS 512KB
put or link it to ./files/thirdparty.bin
A basic build of Sonoff-Tasmota v6.5.0 is already included in this repository.
   # curl http://10.42.42.42/flash3
Alternatively let the device download and flash a file via HTTP:
   # curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin

HAVE FUN!
```
So I continue:
```
root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42/flash2
Device is already booting from userspace 2 (0x81000)
root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
```

Which right now is the current state. On my mobile phone, the `vtrust-flash` network is still there, but no `sonoff-xxx` and it has been like that for a couple of minutes. 5 minutes at least, was I too impatient?
 I can't reach the device as `10.42.42.42` either. Un-Plugging hasn't changed anything either. Right now, it seems to be bricked...

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3689853/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3689854/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3689855/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3689857/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3689858/smarthack-wifi.log)

Any suggestions? Thanks!

kueblc, 2019-10-04:
> ```
> root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42/flash3
> Device should flash http://10.42.42.1/files/thirdparty.bin and restart
> ```

Do you see `thirdparty.bin` being requested in your `smarthack-web.log`?

> Which right now is the current state. On my mobile phone, the `vtrust-flash` network is still there, but no `sonoff-xxx` and it has been like that for a couple of minutes. 5 minutes at least, was I [...]

Definitely not bricked, there's hope!

> Ok, I now have a `vtrust-recovery` SSID, so it seems it successfully flashed the intermediate firmware. Some other comments suggest to restart the flashing (while mobile phone is connected to `vtrust-flash` or `vtrust-recovery`?)

No you do not need to keep your phone connected after the smartconfig process. If you see `vtrust-recovery` this tells you for sure that the device is running the intermediate firmware (not bricked) but you'll need to restart the device (unplug and replug) so that it connects to `vtrust-flash` before being able to do anything else.

> **Ok, next edit**. With the correct sequence of plugging/unplugging and start_flash I was able to get so far again, that I can curl stuff and tried to initiate the flashing again by calling flash3, just like above. Giving it much more time to complete (~30 minutes), I can't still see no `sonoff-xxx` wifi. Unplugging-plugging the switch make me end up again with the `vtrust-recovery` network.

As I said, you can't do much with `vtrust-recovery`, it just lets you know that it still works but can't connect to `vtrust-flash`.

> Trying to initiate flashing by calling
> 
> ```
> # curl "http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin"                                                                                                                                                   
> Device should flash http://10.42.42.1/files/thirdparty.bin and restart
> ```
> 
> instead of just `flash3` gives the same result.

`curl "http://10.42.42.42/flash3"` is the exact same as `curl "http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin"`, it is preferred that you do the first to avoid typos.

How long did you wait between commands?

> How should I proceed from here?

Were you able to get able to get a firmware backup? Please post it if you have one.

floli, 2019-10-04:
> > ```
> > root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42/flash3
> > Device should flash http://10.42.42.1/files/thirdparty.bin and restart
> > ```
> 
> Do you see `thirdparty.bin` being requested in your `smarthack-web.log`?

Yes:
```
[I 191004 19:02:09 web:2246] 200 GET /files/thirdparty.bin (10.42.42.42) 5179.36ms
```
appeared right after calling flash3 (not more than 5 seconds later).

> `curl "http://10.42.42.42/flash3"` is the exact same as `curl "http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin"`, it is preferred that you do the first to avoid typos.
> 
> How long did you wait between commands?

Between which commands?
Between calling flash3 and unplugging without having found a `sonoff-xxx` wifi: ~30 minutes

> > How should I proceed from here?
> 
> Were you able to get able to get a firmware backup? Please post it if you have one.

[firmware-71d2e4.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3691566/firmware-71d2e4.bin.log) (please remove `.log` extension)

Thanks a lot!

kueblc, 2019-10-04:
Unfortunately it looks like that firmware backup did not complete. I won't be able to run this to try reproducing your issue. Let's see if anyone else can comment on their experience with the SP22.

floli, 2019-10-05:
I bought a pair of switches, the other one I left unflashed so far. After several tries I was able to connect it:
```
# curl http://10.42.42.42
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 70de99
MAC: 84:0D:8E:70:DE:99
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```
The firmware from the other, untouched switch, info above: [firmware-70de99.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3692995/firmware-70de99.bin.log)

The firmware from the flashed switch which loads the intermediate firmware and creates the `vtrust-recovery` network: [firmware-71d2e4.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3692996/firmware-71d2e4.bin.log)

Thanks!

Sentinel-sys, 2019-10-05:
Hi, everybody,
I ordered 4 smart sockets type Teckin SP22 from Amazon. The last 2 days I spent flashing these things. But it was only a trial.
I come to the point where the socket should connect to generic Wifi Access Point. But nothing happens except a lot of dots while the shellscript is running and the LED at the socket flashes blue fast and then stops sometime.
In the logfile I can see that my phone has connected to the AP, but the socket doesn't seem to want to connect to the AP.
I read now that there is a new firmware from Tuya which prevents it from flashing. It seems that I already have such a firmware on it. I can exclude an update from the Internet.
I also read that the problem seems to have been fixed with the latest version of Tuya-Convert. If that's true, I can't confirm that.
Is there anything else I can try? My version of Tuya-Convert is from yesterday evening. It runs on a Raspberry Pi with Kali Linux installed.
Thank you

Joe

German:
Hi zusammen,
ich habe mir 4 smarte Steckdosen von Typ Teckin SP22  über Amazon bestellt. Die letzten 2 Tage habe ich damit zugebracht die Dinger zu flashen. Es blieb aber nur beim Versuch.
Ich komme bis zu dem Punkt wo sich die Steckdose mit generieten Wifi Access Point verbinden sollte. Allerdings passiert nichts außer jeder Menge Punkte während das Shellscript läuft und das bei der Steckdose die LED schnell blau blinkt und dann irgendwann aufhört.
Im Logfile kann ich sehen das mein Handy sich mit dem AP verbunden hat, allerdings scheint sich die Steckdose nicht mit dem AP verbinden zu wollen.
Ich habe jetzt gelesen das es wohl eine neue Firmware seitens Tuya gibt, welche das flashen verhindert. Wie es scheint, habe ich eine solche Firmware bereits drauf. Ein Update aus dem Internet kann ich auschliessen.
Außerdem habe ich gelesen das das Problem scheinbar mit der neuesten Version von Tuya-Convert behoben worden ist. Wenn das stimmt, kann ich das so nicht bestätigen.
Gibt es vielleicht noch etwas was ich ausprobieren kann? Mein Version von Tuya-Convert ist von gestern Abend. Sie läuft auf einem Raspberry Pi mit Kali Linux installiert.
Vielen Dank

Joe

kueblc, 2019-10-05:
@floli thanks for the firmware. Did you experience any issues getting flashed?

@Sentinel-sys the new firmware patch has been solved in the latest version of `tuya-convert`. The device MAC address should show up in the wifi log if the smartconfig procedure succeeded. Can you post your logs?

Sentinel-sys, 2019-10-05:
@kueblc thank you for your reply. Here are the latest logfiles

Wifi Log:

`Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan1 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
Using interface wlan1 with hwaddr ca:e0:e0:23:ed:6d and ssid "vtrust-flash"
wlan1: interface state UNINITIALIZED->ENABLED
wlan1: AP-ENABLED 
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: authenticated
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 18:1d:ea:30:68:53
wlan1: STA 18:1d:ea:30:68:53 RADIUS: starting accounting session 24E81877E1D8D1B1
wlan1: STA 18:1d:ea:30:68:53 WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED 18:1d:ea:30:68:53
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: authenticated
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: associated (aid 1)
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: authenticated
wlan1: STA 18:1d:ea:30:68:53 IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED 18:1d:ea:30:68:53
wlan1: STA 18:1d:ea:30:68:53 RADIUS: starting accounting session AE6B82E83D64C67E
wlan1: STA 18:1d:ea:30:68:53 WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED 18:1d:ea:30:68:53
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: associated (aid 1)
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session DDBB85EC4CB0F547
wlan1: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: disassociated due to inactivity
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: did not acknowledge authentication response
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session 5E4124C3BF49C151
wlan1: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session 5E4124C3BF49C151
wlan1: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: associated (aid 1)
wlan1: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session 5E4124C3BF49C151
wlan1: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
wlan1: AP-STA-DISCONNECTED d8:1c:79:7f:58:5b
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: authenticated
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: disassociated due to inactivity
wlan1: STA d8:1c:79:7f:58:5b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`

Web Log:

`    tornado.ioloop.IOLoop.current().start()
  File "/usr/local/lib/python3.6/dist-packages/tornado/platform/asyncio.py", line 148, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 438, in run_forever
    self._run_once()
  File "/usr/lib/python3.6/asyncio/base_events.py", line 1415, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.6/selectors.py", line 445, in select
    fd_event_list = self._epoll.poll(timeout, max_ev)
KeyboardInterrupt`

Do you need also the other logs?

Thank you 

Joe

kueblc, 2019-10-05:
I see stations `18:1d:ea:30:68:53` and `d8:1c:79:7f:58:5b` in your wifi log, is one the device you are trying to flash?

I do not see any requests in the web log. Are there connections in the psk log?

You can create a network capture with `tcpdump -i wlan0 -w capture.pcap` to see what's happening after the device connects.

By the way, you can post your logs with triple backtick as in ` ``` Log here ``` `

Sentinel-sys, 2019-10-06:
@kueblc  
First MAC address is my laptop and second MAC address is my mobile phone. So no smart plug listed there.
I just checked the PSK Log and I didn´t see any connection there.

What I´m trying to do today is, setup a freshly installed Raspberry Pi with Raspbian and try it from there. Then I provide you all the logfiles and pcap files you asked for.

Thank you

Joe

floli, 2019-10-06:
> @floli thanks for the firmware. Did you experience any issues getting flashed?

Haven't tried yet to flash *this* switch, still fear it ends up like the other in the intermediate. Will give it a try later today and update here.

Sentinel-sys, 2019-10-06:
Okay here we go.
New Raspian Buster on a Raspberry 3B+. Latest Updates all installed and newest Tuya Convert loaded from Github.
I followed the intructions and ended with a lot of dots and nothing in the logfiles:

Wifi Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-wifi.log
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:77:09:4e and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA d8:1c:79:7f:58:5b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan0: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session 3507B510F6385066
wlan0: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
```
The MAC address you see here ist my smartphone...

Web Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-web.log
Listening on port 80```

PSK Log: This Logfile is empty

Smartconfig Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-smartconfig.log
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConfig complete.```

MQTT Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-mqtt.log
1570354055: mosquitto version 1.5.7 starting
1570354055: Using default config.
1570354055: Opening ipv4 listen socket on port 1883.
1570354055: Opening ipv6 listen socket on port 1883.```

This time I forgot to do the tcp dump.

I tryed a second time with tcp dump enabled and this is what happend.

Wifi Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-wifi.log
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:77:09:4e and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA d8:1c:79:7f:58:5b IEEE 802.11: disassociated
wlan0: STA d8:1c:79:7f:58:5b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED d8:1c:79:7f:58:5b
wlan0: STA d8:1c:79:7f:58:5b RADIUS: starting accounting session 8DC11DE6BA5B20A8
wlan0: STA d8:1c:79:7f:58:5b WPA: pairwise key handshake completed (RSN)
wlan0: STA 68:57:2d:42:1f:2b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 68:57:2d:42:1f:2b
wlan0: STA 68:57:2d:42:1f:2b RADIUS: starting accounting session 228DCA223FAE9FD0
wlan0: STA 68:57:2d:42:1f:2b WPA: pairwise key handshake completed (RSN)
wlan0: STA 68:57:2d:42:1f:2b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 68:57:2d:42:1f:2b
wlan0: STA 68:57:2d:42:1f:2b IEEE 802.11: disassociated
wlan0: STA 68:57:2d:42:1f:2b IEEE 802.11: disassociated
[... 49 lines cut ...]
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
..........
SmartConf```

MQTT Log:

```pi@raspberrypi:~/tuya-convert/scripts $ tail -f smarthack-mqtt.log
1570354055: mosquitto version 1.5.7 starting
1570354055: Using default config.
1570354055: Opening ipv4 listen socket on port 1883.
1570354055: Opening ipv6 listen socket on port 1883.
^C1570354787: mosquitto version 1.5.7 terminating
1570354794: mosquitto version 1.5.7 starting
1570354794: Using default config.
1570354794: Opening ipv4 listen socket on port 1883.
1570354794: Opening ipv6 listen socket on port 1883.```

How can I upload the TCP dump???

It seems this time I found a device but this is not an ESP32xx device... At the device itself is written "TECKIN SP22"

Thank you

Joe

kueblc, 2019-10-06:
@floli the devices are still stuck in the intermediate firmware? Worst case scenario, they are not bricked, you can connect to `vtrust-recovery` and issue `curl http://10.42.42.42/undo`.

@Sentinel-sys sorry to say that if they are not ESP82xx, there's nothing we can do!

Sentinel-sys, 2019-10-06:
@kueblc 
But is this really true? I can´t imagine that this is a totally different kind of product...?

floli, 2019-10-06:
> @floli the devices are still stuck in the intermediate firmware? Worst case scenario, they are not bricked, you can connect to `vtrust-recovery` and issue `curl http://10.42.42.42/undo`.

I did that and it was confirmed with `Rebooting into userspace 1`. But that's all that changed, the switch still creates the `vtrust-recovery` network after reboot and does not react to button presses.

I can still connect to it using the standard procedure with `start_flash.sh`.
```
root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 71d2e4
MAC: 84:0D:8E:71:D2:E4
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000

root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42/undo
Rebooting into userspace 1

root@marduk ~/tuya-convert (git)-[master] # curl http://10.42.42.42     
[...]
Active Userspace: user2 0x81000
```
(the active userspace has not been changed)

Initiating the flash leaves me at the same intermediate state:
```
 # curl http://10.42.42.42/flash3
Device should flash http://10.42.42.1/files/thirdparty.bin and restart
```

The other, identical, switch I left untouched so far, connected only one to extract the firmware.

Thanks!

Sentinel-sys, 2019-10-06:
So I´m a little bit further. I just cut one of this things in half and had a lock at the PCB. It seems that there is an TYWE2S Chip inside. And this leeds me to this page here: 
[https://github.com/arendst/Sonoff-Tasmota/issues/3950](url)

Maybe there is a way to implement this into Tuya-Convert, so you didn´t need to open these things. It´s quite hard to come in, dough... :-)

Thank you

Joe

kueblc, 2019-10-07:
@Sentinel-sys are you sure it's a TYWE2S and not a WR3? They have started using them in the Teckin smart plugs, even with older boards silk screened with TYWE2S. See #269 

@floli I haven't had the chance to look at your firmware backups. Please follow up with me if I haven't gotten back to you in the next few days.

Sentinel-sys, 2019-10-09:
@kueblc 

I'm afraid I don't really know much about the individual chips. However, on the board next to the corresponding chip is "TYWE2S". I got the plugs from "Gosund" now, at least they still have screws on the backside.
But I can't see any differences in the chips at the moment.
I will take a closer look at it tonight and compare it. I will also try to flash the "Gosund" parts...
Wish me luck

Ich kenne mich leider nicht so wirklich mit den einzelnen Chips aus. Allerdings steht auf der Platine neben dem entsprechenden Chip "TYWE2S". Ich habe mir jetzt einmal die Stecker von "Gosund" besorgt, wenigstens haben die noch Schrauben auf der Rückseite.
Allerdings kann ich aktuell keine Unterschiede bei den Chips feststellen.
Ich werde mir das heute Abend noch einmal genauer ansehen und vergleichen. Außerdem werde ich versuchen die "Gosund" Teile zu flashen...
Wünscht mit Glück

Gruß

Johannes

https://nextcloud.die-dormanns.de/index.php/f/97997
https://nextcloud.die-dormanns.de/index.php/f/98000
https://nextcloud.die-dormanns.de/index.php/f/97987

jjdiazgarcia, 2019-10-09:
I agree with guys that says that new chip is being used by tekin SP22.
Yesterday, I tried to flash them with no luck. Here you are my logs.

```
==> smarthack-web.log <==
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload 593360F8F03D17F6534F121DD6E2090C
**WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware**
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJatKtfL+0oMHlKkGrYoGl499F2VBmqNqTqJcWuiX4YubPyC9VcBmtkwJpq3luO0uBm7mevrGF2WUpAllBDW/Q5l","t":1570554898,"sign":"fca7aa269ae4d49c"}
[I 191008 18:14:58 web:2246] 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=30&uuid=ab712cfef6edfffd&v=4.1&sign=2da154d988bc1829e67a0b37a5015ada (127.0.0.1) 57.42ms
```

I have added bold letters to warning message. It seems they are using a different chip. Will it be supported?

blissarts, 2019-10-09:
Hi,
I was able to flash 4 SP22 sockets (purchased some weeks ago from ebay) with the newest master. I have opened one up and the label is saying TYWE2S.

Host system was an old laptop with Ubuntu 18.04. Had to 
`sudo systemctl disable systemd-resolved`
`sudo systemctl stop systemd-resolved`
to resolve the port 53 issue.
The script is messing with some files in the /etc. Backup the whole folder before starting to be able to get everything back to normal (dns etc.).

Thanks for your effort so far to make this work.

kueblc, 2019-10-09:
@jjdiazgarcia no support is planned for non-ESP82xx chips, as that would require a completely different firmware. I'm not aware of any custom firmware solutions for the Realtek based WR3.

@blissarts these issues should be resolved in the `development` branch, thanks for reporting.

jjdiazgarcia, 2019-10-10:
@kueblc ok. Then, I will send back plugs. Thanks so much.

floli, 2019-10-20:
> @floli I haven't had the chance to look at your firmware backups. Please follow up with me if I haven't gotten back to you in the next few days.

Hey @kueblc ,
were you able to take a look at the firmwares?

Best Thanks!

kueblc, 2019-10-20:
Thanks for following up with me @floli. I just had a look, but they are not complete. A successful backup should be exactly 1MB. How did you create the backups?

floli, 2019-10-21:
I uploaded two firmwares. 

The first one is from an untouched switch, which I haven't tried to flash: [firmware-70de99.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3692995/firmware-70de99.bin.log)

The firmware from the flashed switch which loads the intermediate firmware and creates the `vtrust-recovery` network: [firmware-71d2e4.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3692996/firmware-71d2e4.bin.log).

** old uploads above **

Both firmware files were created from the `start_flash.sh`, right after connecting to the switches.

I fetched the firmware from the semi-flashed (intermediate firmware creating the `vtrust-recovery` network) again:

```
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
  3 1024k    3 32120    0     0  12935      0  0:01:21  0:00:02  0:01:19 12935
curl: Saved to filename 'firmware-71d2e4.bin'
curl: (18) transfer closed with 1016456 bytes remaining to read
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.1
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 71d2e4
MAC: 84:0D:8E:71:D2:E4
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 144051
FlashChipRealSize: 1024K
Active Userspace: user2 0x81000
```
[firmware-71d2e4.bin.log](https://github.com/ct-Open-Source/tuya-convert/files/3752287/firmware-71d2e4.bin.log) (filesize is only 32K).

From the other, unflashed device I wasn't able to get another firmware images. Connecting to it seems to be a game of luck, without much luck on my side. As described above, it goes back from pairing mode to normal when `Waiting for the upgraded device to appear`.

floli, 2019-11-11:
Hey,

with the most recent version of tuya convert I was able to successfully flash both devices.

Thanks for your help!

closed 2019-11-11 by floli

kueblc, 2019-11-11:
Great! Thanks for following up!

henfri, 2019-11-26:
Hello,

for me, even the latest version of tuya convert reports that no ESP82xx is contained in my SP22 (shipped yesterday from Amazon (Fanglishi EU).

Greetings,
Hendrik

blohm19, 2019-11-30:
Hello,

same for me. Two Teckin SP22 arrived yesterday and I get for bot the error, that no ESP82xx is contained.

kueblc, 2019-11-30:
Yes @henfri and @blohm19, `tuya-convert` will report this for your benefit, so that you understand why the flash is not successful and so that you do not attempt to flash the device by serial.

Do not expect this to change with any updates to `tuya-convert` as of course `tuya-convert` can not modify your device's hardware.

henfri, 2019-11-30:
I know.
We are trying to help others by building some statistics.
Tuya-convert is working Well. Just the Message about the wrong hardware could bei written in the Main Screen instead of being somewhat hidden in the logs.

kueblc, 2019-11-30:
OK great just making sure. Thanks for the feedback!

henfri, 2019-11-30:
Yes, thank you. I think your clarification is useful