ct-Open-Source/tuya-convert #286: Tekin SP21 - unable to flash even with HTTPS merged

gbukovszki, 2019-10-03:
Flashing environment: RasPi  3 Model B v1.2
Device is a Tekin SP21 with the Firmware 1.0.2 factory installed
tuya-convert cloned freshly on 03.10.19

Logs attached.
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3686937/smarthack-web.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3686938/smarthack-smartconfig.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3686939/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3686940/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3686941/smarthack-wifi.log)

kueblc, 2019-10-04:
Try #289

kueblc, 2019-10-04:
Should be fixed in `master` now, let us know if you have any further questions or issues

closed 2019-10-04 by kueblc

gbukovszki, 2019-10-05:
Tested, works. Many thanks guys! (Any donation possibility? Where should I send the beer?)

kueblc, 2019-10-05:
Awesome @gbukovszki. Thank you for the offer, we do not have an official tip jar set up yet, but if you'd like, you could [help me save my cat](https://www.gofundme.com/f/figgy039s-breast-cancer-recovery-fund) :smile_cat: