ct-Open-Source/tuya-convert #287: Internet Unavailable on vtrust-flash

kloggy, 2019-10-03:
Hi, the subject says it all really.

I can connect to the vtrust-flash network ok with my Android phone but I get the message, 'The Internet Is Unavailable' despite several reboots of the RPi. My RPi can ping the outside so there is a connection there somewhere. I'm happy to provide more info if needed.
Thanks for any suggestions!

kueblc, 2019-10-03:
As intended, there shouldn't be internet on the `vtrust-flash` AP. It is intended to be isolated.

closed 2019-10-03 by kueblc

johny-mnemonic, 2020-01-17:
@kueblc  I understand why are you doing this, but it is almost impossible to convince Android phone to stay connected to Wifi network that doesn't have internet and doesn't use any sort of security. It avoids it like a plague (disconnects immediately).
Do you know about any trick to convince Android phone to stay connected to such network?

GeorgeIoak, 2020-01-17:
If it's any help, what I do is select the `vtrust-flash` network from the WiFi page and then immediately go back to the home screen. I find that if I do that I get the pop up message asking me to confirm that I want to connect to this network. Once I do that my phone seems to stay connected without any issues. Sometimes I see a message saying that I need to login to `vtrust-flash` but if I click on that a web page opens saying that I already am logged in.

Another option is use Digiblur's code on another ESP device (say a Wemos D1 Mini) that will connect and stay connected to the `vtrust-flash` AP. You can find that at https://github.com/digiblur/Tuya-Convert-Donor