ct-Open-Source/tuya-convert #288: Support for BlitzWolf® BW-SHP8

WladyX, 2019-10-03:
Hello,
I've got a BW SHP8:
https://www.banggood.com/BlitzWolf-BW-SHP8-3680W-16A-Smart-WIFI-Wall-Outlet-EU-Plug-Socket-Timer-Remote-Control-Power-Monitor-Work-with-Alexa-Google-Assistant-IFTTT-p-1502402.html?cur_warehouse=CN

I've tried to flash it, but the processed was unsucessfull, can you please tell me if this is a supported socket?
Thank you very much for all your work!

`wlan0: AP-STA-CONNECTED dc:4f:22:b3:63:35
wlan0: STA dc:4f:22:b3:63:35 RADIUS: starting accounting session 37343DB6C8EE41D5
wlan0: STA dc:4f:22:b3:63:35 WPA: pairwise key handshake completed (RSN)
and then it just hangs, the led is blue stable - so prob it is connected to the AP`

kueblc, 2019-10-03:
Please post your logs

kueblc, 2019-10-03:
[Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting)

WladyX, 2019-10-03:
Added bellow, thanks!

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3687639/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3687640/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3687641/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3687642/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3687643/smarthack-wifi.log)

WladyX, 2019-10-03:
I looked at the logs, but not very closely it seems, once I uploaded them here I thought I should take another look and then I saw a missing sslpsk module, so I ran the install prereq script (which I could have swore I ran before) and tried to flash and all worked as expected.
So problem was between the chair and the keyboard.
You can mark this device as supported.
Thank you @kueblc !

closed 2019-10-03 by WladyX