ct-Open-Source/tuya-convert #290: Tekin Smart Plug SP27 - Possible to flash?

kloggy, 2019-10-04:
The SP27 Smart Plug (UK) does not appear in the Wiki.

Using the latest tuya-convert from today (04/10/19 @ 10am).

I get as far as "Waiting for the upgraded device to appear', with the dots going across the screen but they just keep going. I left it a few minutes, but no progress. It didn't proceed to the 'IoT device is on-line with ip x.x.x.x' message.

I think that I've read that some devices now have different WiFi chips making them impossible to flash. Could this be my problem? Do I need to return them to Amazon :-) ?

Thankyou for any help/guidance

LOGS:

smarthack-psk.log
EMPTY

smarthack-web.log
```
Listening on port 80
```

smarthack-mqtt.log
```
1570178630: mosquitto version 1.5.7 starting
1570178630: Using default config.
1570178630: Opening ipv4 listen socket on port 1883.
1570178630: Opening ipv6 listen socket on port 1883.
```

smarthack-smartconfig.log
```
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:xx:xx:xx and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: STA a8:b8:6e:xx:xx:xx IEEE 802.11: associated
wlan0: AP-STA-CONNECTED a8:b8:6e:xx:xx:xx
wlan0: STA a8:b8:6e:xx:xx:xx RADIUS: starting accounting session 7D40C7xxxxxx87D9
wlan0: STA a8:b8:6e:xx:xx:xx WPA: pairwise key handshake completed (RSN)
wlan0: STA a8:b8:6e:xx:xx:xx IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED a8:b8:6e:xx:xx:xx
wlan0: STA a8:b8:6e:xx:xx:xx IEEE 802.11: disassociated
```

smarthack-wifi.log
```
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
```

kueblc, 2019-10-04:
Looks like the device never even connected. Was the device in EZ config mode (fast blink) or AP config mode (slow blink)? Did it stop blinking at any point? Is there an FCC ID?

kueblc, 2019-10-04:
[The manufacturer claims](https://fccid.io/2AO3S-SP20/Letter/Family-Models-Declaration-Letter-3765504) that it should be the same hardware as the rest of the SP** series

kloggy, 2019-10-04:
Fast blink here and it never stopped blinking.
There is no FCC ID, this is a UK device, isn't FCC specific to the US?
There is a barcode (on the box of a two pack) if that helps?
X0011EW0AF

Let me know if there is anything I can provide...

kloggy, 2019-10-04:
I should have said the barcode is a sticker which would imply some kind of uniqueness as the box itself is printed with the model SP27

kueblc, 2019-10-04:
> Fast blink here and it never stopped blinking.

OK right mode, so something else is going wrong

> There is no FCC ID, this is a UK device, isn't FCC specific to the US?

Yes it is specific to US, but the Amazon page mentioned FCC approval and since they use the same hardware internationally I thought it might tell us what said hardware is. The link I posted earlier is from my search through FCC records.

> There is a barcode (on the box of a two pack) if that helps?
> X0011EW0AF
>
> The barcode is a sticker which would imply some kind of uniqueness as the box itself is printed with the model SP27.

Could be that this is a totally different module.

Would you be able to capture network traffic while pairing the device with the official app?

You would need to set up a hotspot on your machine, run `tcpdump` or WireShark on the access point interface, and then connect your phone before using the app to pair the device.

kloggy, 2019-10-04:
> Would you be able to capture network traffic while pairing the device with the official app?
You would need to set up a hotspot on your machine, run tcpdump or WireShark on the access point interface, and then connect your phone before using the app to pair the device.

Ok, I understand what you are asking but achieving it is right at the edge of my knowledge. I am not at all proficient in Linux. I'll have a read around and see what I can do.

ceaswaran, 2019-10-05:
Am also suffering from same issue. Plug NX-SM223 Pressing button 6s makes it to go to fastbliniking mode. (Working fine)
No, it did not stop blinking at any point after connecting to Vtrust wifi

`wlan0: AP-ENABLED 
wlan0: STA 08:25:25:95:68:8b IEEE 802.11: associated
wlan0: AP-STA-CONNECTED 08:25:25:95:68:8b
wlan0: STA 08:25:25:95:68:8b RADIUS: starting accounting session 2C6DC076165E7189
wlan0: STA 08:25:25:95:68:8b WPA: pairwise key handshake completed (RSN)
wlan0: STA 08:25:25:95:68:8b IEEE 802.11: disassociated
wlan0: AP-STA-DISCONNECTED 08:25:25:95:68:8b
wlan0: STA 08:25:25:95:68:8b IEEE 802.11: disassociated
`
Additional Info: I have flashed the same plug a few months ago breaking it open & hard wiring FTDI flash. It uses BLITZWOLF SHP module & tasmoto & works great.

Please help

ceaswaran, 2019-10-05:
**How it worked for me finally:**

Not sure if this work for all but I tried the following & Tuya convert worked for my 16A Plug NX-SM223 
(I ASSUME ITS BLITZWOLF SPH CLONE OR REBRANDED)

Update: 
I tried to configure with a tuya cloud app, it requested firmware update in which I updated the firmware.
After which I gave the factory rest device in tuya cloud app.
Then tried tuya convert using this github link https://github.com/M4dmartig4n/tuya-convert

It worked & I converted my plug to Tasmota Firmware.

You can give it a try.  I have no idea why it worked & other methods dint.

fyi

kloggy, 2019-10-05:
Thanks for the update.

I'm interested if @kueblc has a view on this before I go ahead and try? I'd rather not risk bricking my device trying something that seems a bit 'random' with no understanding of why it worked, especially as my plug is a different model to you (13A for starters).

Obviously it is great news that it did work for you!

kueblc, 2019-10-05:
There is no difference between the smart config procedure in @M4dmartig4n's fork and `master`. And now that the fork has been merged there should be no difference whatsoever. I don't recommend using the older fork as it's no longer being maintained.

I'm curious if the firmware update made a difference, but there are so many different devices with different firmware, it's difficult to test.

Another possible problem you both may have encountered is getting the timing right for the smartconfig process. The `start_flash` script tells you to connect a phone and then press enter, at that point it'll broadcast the config packets for 5-10 seconds. If it didn't connect in that time, it won't have another opportunity to.

#276 aims to fix this by repeating the config process several times before giving up. You could try this branch, or simply try repeating the smartconfig manually, by issuing `./smartconfig/main.py` from the `scripts` folder, after putting your device in pairing mode.

yabolek, 2019-10-05:
Would you be so kind and have a look at my logs? Blitzwolf BW-SHP6

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3693580/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3693581/smarthack-web.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3693582/smarthack-smartconfig.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3693583/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3693584/smarthack-mqtt.log)

kueblc, 2019-10-05:
@yabolek 

psk log has `OSError: [Errno 98] Address already in use` meaning you have another service listening on port 443 or 8886. See if you can find out what's occupying those ports and terminate the process if you can. `netstat` ought to help with that.

wifi log has `dnsmasq: failed to create listening socket for port 53: Address already in use`, so you can use the above strategy or try out #225 

Also I only see one mac address in the wifi log, I'm assuming this is your phone, and pairing never succeeded. You might try running `./smartconfig/main.py` several times to get the device to pick up the configuration. This is also staged for #276

kloggy, 2019-10-07:
@kueblc 

I tried running `./smartconfig/main.py` several times but still no luck...

```
 pi@raspberrypi:~/tuya-convert/scripts $ ./smartconfig/main.py
Put Device in Learn Mode! Sending SmartConfig Packets now
Sending SSID                  vtrust-flash
Sending wifiPassword          flashmeifyoucan
SmartConfig in progress
^CTraceback (most recent call last):
  File "./smartconfig/main.py", line 23, in <module>
    smartconfig( passwd, ssid, region, token, secret )
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 54, in smartconfig
    sock.send_multicast(multicast_head)
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 39, in send_multicast
    self._socket.sendto( b'\0', (ip, 30012))
KeyboardInterrupt
```

The plug stayed in fast blinking mode throughout.

Are the logs useful? If so I will post them but can you tell me if there is anything in them I should redact first?

kueblc, 2019-10-07:
@kloggy if the device did not stop blinking there is probably nothing in the logs. You can check the wifi log to see if the device's MAC address appeared.

kloggy, 2019-10-07:
No it doesn't appear, just the MAC of the Pi and the phone I used to connect to 'vtrust-flash'.

Is it worth pursuing or should I give up with these plugs for now and return to Amazon? :-(

kueblc, 2019-10-08:
Not sure what would be preventing the smart config process from continuing. Have you been able to flash any other devices with this set up?

kloggy, 2019-10-08:
Not tried, this is my first go with any devices.

kloggy, 2019-10-08:
@kueblc 
Thanks for all your efforts, not just with me but the whole thing. For now though I'm afraid I'm going to have to give up and send back my plugs. Hopefully I can try again another time. I'll be watching for any further progress here. As I said already, thank you I appreciate the work you've put in.

kueblc, 2019-10-09:
You're welcome, it's a shame we couldn't get it working. Feel free to follow up if you have any other questions or issues.

closed 2019-10-09 by kueblc

yabolek, 2019-11-26:
> @yabolek
> 
> psk log has `OSError: [Errno 98] Address already in use` meaning you have another service listening on port 443 or 8886. See if you can find out what's occupying those ports and terminate the process if you can. `netstat` ought to help with that. **EDIT:** #225 might solve this too
> 
> wifi log has `dnsmasq: failed to create listening socket for port 53: Address already in use`, so you can use the above strategy or try out #225
> 
> Also I only see one mac address in the wifi log, I'm assuming this is your phone, and pairing never succeeded. You might try running `./smartconfig/main.py` several times to get the device to pick up the configuration. This is also staged for #276

Hello @kueblc ,

Thanks so much for looking into my issue. Today I got around flashing all of my plugs and it worked perfectly! I must say that you are making a hell of a job both developing and supporting this tool. It would be a pleasure for me to support your work. People like you are making this world a better place. Thanks again!

kueblc, 2019-11-26:
That's great @yabolek, I'm really happy to hear it went smoothly this time!

If you are looking for ways to support the project, you may consider becoming a financial sponsor to help offset my time and hardware expenses. You can also contribute to development, testing PRs, and spreading the word to grow the community!

Very kind of you to write in, happy to hear my work is helping people :)