ct-Open-Source/tuya-convert #293: Can't get AP to launch. Tried Kali, Ubuntu, and pi 3 B

scy0846, 2019-10-04:
I have tried getting this AP to launch in 3 separate devices. In all of them it turns the AP on for a second and then turns it immediately off. Only enough to discover, but not connect to. I first tried in Ubuntu and when it wouldn't work  I tried a fresh install of kali on three separate computers and couldn't get those to work. I then noticed most people on YouTube are using a pi for this and it seems like it will give me the least amount of issues. I purchased a brand new raspberry pi 3 version B, bought a brand new sd card and installed prereqs. I then launched the program script and it just shows it's switching to AP, but immediately turns off and connects back to my wifi router. Any ideas?

kueblc, 2019-10-04:
Do you want to try #225?
```
git clone -b fix-net-inf https://github.com/kueblc/tuya-convert.git
```

pfeerick, 2019-10-04:
> but immediately turns off and connects back to my wifi router. 

Um, are you using the WiFi to connect to the Raspberry Pi? I didn't encounter this issue when using either that latest versions of Raspbian Stretch or Buster, but I also didn't configure the Wireless at all and used the hard-wired Ethernet connection with it.

scy0846, 2019-10-05:
@kueblc Looks like that fixed it. I can't get my light to flash but they are brand new. Maybe Tuya patched it. Regardless, This can be closed.

closed 2019-10-05 by scy0846

kueblc, 2019-10-05:
@scy0846 can you post your logs?

scy0846, 2019-10-07:
sure. I will be sure to do that later tonight. I was able to flash one bulb from another company, but the bulbs I can't get flashed with this method are these: https://www.amazon.com/gp/product/B07VP2KL7W

I'm really just thinking that I can't flash them because maybe Tuya patched it or something. Regardless, I appreciate the work you guys have done on this.

scy0846, 2019-10-07:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3699289/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3699290/smarthack-psk.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3699291/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3699292/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3699293/smarthack-wifi.log)

kueblc, 2019-10-07:
As far as I know, Tuya has not yet released a patch for the latest `tuya-convert`. It does look like there was a successful flash in your web log. Was that the other bulb you spoke of?

scy0846, 2019-10-07:
Good news. I got the latest development branch and that flashed the bulb beautifully. Thanks for your help. Awesome project 

⁣Sent from BlueMail ​

kueblc, 2019-10-07:
Great, happy to hear it!