ct-Open-Source/tuya-convert #298: Running tuya-convert on a Mac?

tonysprenk, 2019-10-06:
Is it possible to run Tuya-convert on a Mac? I've tried Kali Linux in VMware Fusion but I haven't been able to figure out how I can get it to use the built in wifi card. 

Any suggestions?

Thanks allot and thanks for all the hard work put into this.

fastcolors, 2019-10-07:
not possible, you'd need an external WIFI dongle. built-in Mac wifi card won't let linux create an ad hoc network (although osx can). At least this is what I found out trying to run tuna-convert on my Macbook pro late 2013.

kueblc, 2019-10-29:
Closing as question has been answered, please feel free to comment with any updates

closed 2019-10-29 by kueblc

phixion, 2019-12-06:
been using tuya convert on 2 macbooks  both have the broadcom43 both run arch (i recall using b43fwcutter on at least one of them) and always worked flawless.

fastcolors, 2019-12-07:
> been using tuya convert on 2 macbooks both have the broadcom43 both run arch (i recall using b43fwcutter on at least one of them) and always worked flawless.

Using only the built in WiFi? Could you share how?

phixion, 2019-12-09:
to be honest I didnt do anything special but here's my setup and my modified install_prereq.sh to make it work on arch. I ran it on a newer model aswell (however both models are ancient 3,1 and 6,1 macbook air)

https://gist.github.com/phixion/f9bf065bb8970bdda3d35d3842438780

I guess it comes down to the wifi adapter the macbook has, those run a broadcom 43xx

godpurpose, 2020-07-04:
> to be honest I didnt do anything special but here's my setup and my modified install_prereq.sh to make it work on arch. I ran it on a newer model aswell (however both models are ancient 3,1 and 6,1 [...]

Could you please share how to do it??? I've been trying different ways with no success.
I'd really appreciate it.
I have the Broadcom 43xx wifi adapter.