ct-Open-Source/tuya-convert #300: Can't connect to AP

dark-nova, 2019-10-07:
Hi,

I have not been able to get past the "connect device to AP" step. I have tried setting up on two machines: an Asus laptop on Ubuntu 18.04 and a single board computer on Lubuntu 18.04. I tried connecting using a Chromebook, a phone running LineageOS, and a phone running Ubuntu Touch. Unfortunately, my attempts to connect were in vain. The Chromebook mentioned a DHCP configuration issue on the AP, while both phones endlessly attempted to connect.

Is there something else I should look at?

Thanks for your time.

kueblc, 2019-10-07:
Is this on `master` branch or `development`? Try the other one

dark-nova, 2019-10-07:
Thank you, that helped. I switched from `master` to `development` and everything worked.

closed 2019-10-07 by dark-nova

dark-nova, 2019-10-07:
On a side note, since the flash worked, I will try to add to the wiki (AUKEY Wi-Fi Smart Plug) but I don't know whether I am filling in correctly, and I definitely don't know about some columns.

kueblc, 2019-10-07:
Awesome, glad to hear it worked for you. Don't worry about filling in all the columns, just add the info you have and leave the rest blank.