ct-Open-Source/tuya-convert #301: Tuya-Convert: WiFi doesn't appear

carl2298, 2019-10-07:
I want to flash one of these chinese power strips with the help of tuya-convert but the WiFi doesn't appear. Because I don't have a Raspberry Pi I tried it with the Windows subsystem Ubuntu. And I'm using a USB-WiFi-Stick on my computer but normally that should work. Can anyone help me solve this issue? Or did anyone tried it with a subsystem and had success?

scy0846, 2019-10-07:
try this. It worked for me

`git clone -b fix-net-inf https://github.com/kueblc/tuya-convert.git`

carl2298, 2019-10-08:
Thank you. But it's still not working.

scy0846, 2019-10-08:
Try using the development branch. It worked well for another issue I was
having but there may be a fix in it that will help you

WaaZaa666, 2019-10-13:
I tried using WSL Ubuntu to no avail. I now have a Kali live USB, which I run to do the flashing..

kueblc, 2019-10-17:
@carl2298 did you resolve this? Have you tried the `development` branch?

carlzech, 2019-10-19:
I'm now trying it with a Raspberry Pi and a new WiFi-Stick. The WiFi does apppear but everytime I try to connect to the WiFi with my Smartphone the tool shows a lot of new lines and then the Raspberry Pi hang up. And I have to restart it because it's not responding to anything. I thought I made a mistake anywhere so I tried everthing again but I still have the same issue. The last two lines before it hang up: "Kernel panic - not syncing: Fatal exception in interrupt //// ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---" 
Can anyone help me with that?

CaffeinatedFunctionality, 2019-10-19:
Did you use the development branch though? It may help

⁣Sent from BlueMail ​

carlzech, 2019-10-19:
How do I use the development branch?

cedricgrothues, 2019-10-20:
git clone -b development https://github.com/ct-Open-Source/tuya-convert/

carlzech, 2019-10-21:
I now tried it with the development branch but it's still not working. Same problem... I'm using a Raspberry Pi 1 may this be a problem?

carlzech, 2019-10-21:
I now tried it with the development branch but it's still not working. Same problem... I'm using a Raspberry Pi 1 could that be a problem?

ivooovi, 2019-10-21:
I have the same problem although in the past it worked fine. But what I do is before running the ./start_flash command I do : sudo killall wpa_supplicant
This resolved it for me

kueblc, 2019-10-21:
Please post your logs

kueblc, 2019-10-29:
Closing while we wait for author to respond

closed 2019-10-29 by kueblc

Bobofet682, 2019-11-04:
I have the same issue like carlzech.
After WLAN connect -> Kernel Panic and raspberry shows a lot of new lines and then the Raspberry hang up. The Raspberry wont responding to anything. No Keyboard input, no ping, no ssh, nothing.

What I have tested:
As well Developement as Master Branch.
New installed and Updated (and upgraded) Raspbian Buster Lite.
3 different USB WLAN Sticks tested.
Raspberry 2 Model B V1.1 (As well tested with Raspberry 1 Modell B+)

sudo killall wpa_supplicant does not fix the Problem

What can i do to help fixing the problem?

carlzech, 2019-11-08:
Sorry for not responding so long. I was busy and didn't had the time to test anything. I now tried what @Bobofet682 said in the linked issue but it's still not working. Same issue as before...