ct-Open-Source/tuya-convert #302: Trying to flash Jisusu SP-111

zzzZZZzzz9, 2019-10-07:
Trying to use tuya convert on a clean RPi3 Raspbian Buster with a Jisusu SP-111. Freeware according to Smart Life app is 1.0.0.
All installations work fine, AP is created and device connected to it. As soon as I start the flash, the device leaves pairing mode and turns on (red light). I have repeated this process a few times now with the same result. Any ideas?

Attached are the logs
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3699108/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3699109/smarthack-mqtt.log)

[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3699110/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3699111/smarthack-web.log)

manju-rn, 2019-10-08:
Is your network disconnected to internet?  try doing this process when not connected to internet. I have been able to flash this

kueblc, 2019-10-08:
These logs indicate you are using an outdated `tuya-convert`, update and try again, let us know how it goes.

zzzZZZzzz9, 2019-10-08:
I just set up the system yesterday. Did the download link change? Not sure how it can be outdated otherwise...

kueblc, 2019-10-08:
Not sure what download link you're referring to, you should install by running
`git clone https://github.com/ct-Open-Source/tuya-convert.git` or
`git clone -b development https://github.com/ct-Open-Source/tuya-convert.git`

kueblc, 2019-10-17:
@zzzZZZzzz9 is this resolved? If so, please close the issue

closed 2019-10-29 by kueblc