ct-Open-Source/tuya-convert #303: Driver does not support authentication/association or connect commands

tonysprenk, 2019-10-09:
I'm running Kali Linux latest version in VMware fusion on a macbook pro. The VM is setup to have a "wired" connection to the host macbook pro which is connected to the internet through ethernet. I have a Realtek wifi dongle connected to the VM. This is my wifi log:

```
Backing up NetworkManager.cfg...
Restarting NetworkManager...
Backing up /etc/dnsmasq.conf...
Writing dnsmasq config file...
Creating new /etc/dnsmasq.conf...
Writing hostapd config file...
Configuring AP interface...
Applying iptables rules...
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
```

Any help would be greatly appreciated. And thanks to everyone for making all this possible!

tonysprenk, 2019-10-09:
I tried the create_ap branch and now the log looks like this:
```
WARN: Your adapter does not fully support AP virtual interface, enabling --no-virt
Config dir: /tmp/create_ap.wlan0.conf.eIiHAGxt
PID: 28492
Network Manager found, set wlan0 as unmanaged device... DONE
No Internet sharing
hostapd command-line interface: hostapd_cli -p /tmp/create_ap.wlan0.conf.eIiHAGxt/hostapd_ctrl
Configuration file: /tmp/create_ap.wlan0.conf.eIiHAGxt/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started

Error: Failed to run hostapd, maybe a program is interfering.
If an error like 'n80211: Could not configure driver mode' was thrown
try running the following before starting create_ap:
    nmcli r wifi off
    rfkill unblock wlan

Doing cleanup.. done
```

kueblc, 2019-10-09:
I'm not sure if this would help, I don't have a Mac to test on, but you could try the `development` branch which has a number of net interface fixes.

dmascord, 2019-10-15:
It is probable that your wifi driver for the USB dongle doesn't support AP mode out of the box.

I would try to get the latest possible driver, such as https://github.com/lwfinger/rtl8188eu and see if that resolves the issue.

kueblc, 2019-10-29:
Seems like support for Mac is not possible at the moment, but please feel free to update us with any findings

closed 2019-10-29 by kueblc