ct-Open-Source/tuya-convert #305: Teckin SP20 First flash - No LED no AP

Extarys, 2019-10-09:
Fresh Rasbian build
I ran
```
sudo apt update && sudo apt upgrade
sudo apt dist-upgrade
```
then I clone this repo, ran the prereq script and started the process.
I typed `curl http://10.42.42.42/flash3` and waited 10 minutes. No AP was created, the LED was off. I unplugged/plugged the device and now it blinks red/blue 1 time at boot and then blue once. Then no LED.

I saw some python error in the logs :cry: 

Teckin SP20 (US)

Logs:
https://transfer.sh/gdOGS/smarthack-wifi.log
https://transfer.sh/Hd2RM/smarthack-web.log
https://transfer.sh/C0yHH/smarthack-mqtt.log

kueblc, 2019-10-09:
It does look like the device flashed the intermediate firmware. While the `vtrust-flash` SSID is available, plug the device in. When you see it connect in the wifi log, try to do `curl http://10.42.42.42`. If that succeeds, try to do `curl -JO http://10.42.42.42/backup`. If that succeeds, go on to `curl http://10.42.42.42/flash3`.

Extarys, 2019-10-10:
@kueblc The device never connect to the AP. :disappointed:

meingraham, 2019-10-10:
@Extarys 

What Colin described is prior to the device being flashed with the custom firmware (i.e., Tasmota). The "intermediate" firmware is just that, the step before the custom firmware is loaded on the device.

- ./start_flash - to initialize the Raspberry Pi to start up the vtrust-flash AP
- Connect a local device (e.g., your mobile phone) to vtrust-flash
- Bring up another terminal window connected to your Raspberry Pi and `tail` the wifi log
- Power up your SP20 and watch the `tail` terminal to see if it connects
- If it does, you can stop the `tail` (ctrl-c)
- Run `curl http://10.42.42.42`
- If that is successful, start the firmware flash - `http://10.42.42.42/flash3`

If this is successful, you should then see the `sonoff-xxxx` wireless AP show up.

Mike

P.S. I flashed 10 AWP04L tonight with no issues. These are the same device as the SP20.

MwC-Trexx, 2019-10-11:
I am running into a similar issue with the new sw stack re: SP20.   I bought a box of 4 several months ago, 2 flashed via the older tuya-convert version.  2 won't flash with either old code or new dev code branch. 

Neither one is connecting to vtrust AP.  I have tried to get them into "pairing mode" but they just won't do it.  Holding button while plugging in, waiting and then holding button, etc.

I am tempted to try loading the app on a phone to see can get 1 of them into right "mode" to connect.  

Any suggestions???

kueblc, 2019-10-11:
You can try to capture network traffic by connecting both your phone and device to the `vtrust-flash` AP and recording with WireShark or `tcpdump`.

MwC-Trexx, 2019-10-11:
@kueblc The device isn't connecting to the vtrust-flash which is the issue.  I even tried connecting via smartlife app to no avail.   I sent Teckin support an email to see if I can get the 2 devices swapped, so will see what happens.

Extarys, 2019-10-23:
Tried flashing my second one today, I also broke it. I'll try to replace them through Amazon.

```
1571860572: New client connected from 127.0.0.1 as 44ca41a4-25bb-45f3-a78a-db836b0605dc (c1, k60).
1571860572: No will message specified.
1571860572: Sending CONNACK to 44ca41a4-25bb-45f3-a78a-db836b0605dc (0, 0)
1571860572: Received PUBLISH from 44ca41a4-25bb-45f3-a78a-db836b0605dc (d0, q0, r0, m0, 'smart/device/in/23027313bcddc2923ee5', ... (147 bytes))
1571860572: Sending PUBLISH to 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/in/23027313bcddc2923ee5', ... (147 bytes))
1571860572: Received DISCONNECT from 44ca41a4-25bb-45f3-a78a-db836b0605dc
1571860572: Client 44ca41a4-25bb-45f3-a78a-db836b0605dc disconnected.
1571860573: Received PUBLISH from 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/out/23027313bcddc2923ee5', ... (80 bytes))
1571860573: Received PUBLISH from 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/out/23027313bcddc2923ee5', ... (80 bytes))
1571860574: Received PUBLISH from 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/out/23027313bcddc2923ee5', ... (81 bytes))
1571860578: Received PUBLISH from 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/out/23027313bcddc2923ee5', ... (80 bytes))
1571860579: Received PUBLISH from 23027313bcddc2923ee5 (d0, q0, r0, m0, 'smart/device/out/23027313bcddc2923ee5', ... (81 bytes))
1571860624: Client 23027313bcddc2923ee5 has exceeded timeout, disconnecting.
1571860624: Socket error on client 23027313bcddc2923ee5, disconnecting.
1571861141: mosquitto version 1.5.7 terminating
```

I'd like to try slitinto, which is also tuya, no idea if it would work. 
Is there another plug I could order that is somehow better for flashing OTA?

kueblc, 2019-10-23:
It is not clear what you mean by broke it. The log you post is entirely normal.

Extarys, 2019-10-23:
Oh well sorry haha. Same as the first one. I plug it in, blue blink once. That's it. The "never-ending dots" never ended. I never saw the device with fing.

```
Starting pairing procedure in screen
RTNETLINK answers: File exists
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................```

What is the preferred way to upload my logs?

kueblc, 2019-10-23:
You can copy and paste them into a post between triple backticks (` ``` `) or attach as files. The web log is usually the most helpful for debugging.

Extarys, 2019-10-23:
I love scp.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3764483/smarthack-mqtt.log)
[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3764484/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3764485/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3764486/smarthack-wifi.log)

I hope I just messed up, so I can just order more and flash them properly :)

kueblc, 2019-10-23:
Do you see a `vtrust-recovery` SSID from the plug?

Extarys, 2019-10-23:
No, my cellphone doesn't see any wifi access point except my neighbours and mines.

kueblc, 2019-10-23:
No it wouldn't be hidden. You may have to wait 30 seconds or so after plugging the device in, and it would only produce `vtrust-recovery` if it can't find `vtrust-flash`, and this is if the device is indeed running the intermediate firmware. From the most recent logs you posted, it looks like it should be.

Extarys, 2019-10-23:
Oh really? It didn't say it flashed on the script. I'll let it plugged longer. I'll report back soon.
Thanks for helping me.

Extarys, 2019-10-23:
There is no access point from neither of the SP20 I have. They may have patched the vulnerability allowing for OTA? I'm currently reading on ESPHome too, maybe I could try that, but I don't understand anything about that esphome thing lol.

kueblc, 2019-10-23:
Your logs indicate the hack was successful, so it's not that it's patched, but sounds like something went wrong after installing the intermediate firmware. It's hard to say what without having the serial output from the device.

Have you tried starting `start_flash`, then plugging the device back in and waiting a minute? (No need to enter pairing mode or connect a phone)

Extarys, 2019-10-24:
I tried as you said. There is no activity. No connection in the wifi logs either.

They probably patched the vulnerability or something. Is it guarantee that I can flash it if I had a serial-usb cable and a solder iron? I could maybe try and get one. Never did that though.

kueblc, 2019-10-24:
You should be able to flash it, but I can't make any guarantees. If you do manage to wire one up, make a firmware backup before flashing and we can use that to debug what went wrong.

kueblc, 2019-11-10:
@Extarys can this be closed or is there anything else you need help with?

kueblc, 2019-11-17:
Closing due to inactivity, please feel free to reopen if there are further concerns

closed 2019-11-17 by kueblc