ct-Open-Source/tuya-convert #306: Cant connect to Vtrust wifi on RPI2

xmatamoskax, 2019-10-09:
I have gone through the tutorial 2 different times fresh installs and get up to connecting to the vtrust wifi but i get incorrect wifi password every time with different devices (iphone, android, and laptop).

kueblc, 2019-10-09:
Not enough info to help you, other than to triple check you are indeed using the correct password `flashmeifyoucan` and try the Troubleshooting page on the wiki. If you are still having trouble please follow the instructions when opening an issue and post your logs.

closed 2019-10-09 by kueblc

xmatamoskax, 2019-10-09:
> Not enough info to help you, other than to triple check you are indeed using the correct password `flashmeifyoucan` and try the Troubleshooting page on the wiki. If you are still having trouble please follow the instructions when opening an issue and post your logs.

password is correct and there isnt much on the wiki for this issue. How can i pull a log on the ap?