ct-Open-Source/tuya-convert #307: Won’t make ap

alunarkid, 2019-10-10:
Hello my raspberry pi won’t make the ap I will grab some logs for it all

kueblc, 2019-10-10:
Try `development` branch and please answer all questions when opening an issue

alunarkid, 2019-10-10:
==> ./scripts/smarthack-wifi.log <==
Starting DNSMASQ server...
Starting AP on wlan0 in screen terminal...
Configuration file: /etc/hostapd/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started

alunarkid, 2019-10-10:
Sorry I’m still new to all this how would I use the development branch

kueblc, 2019-10-10:
Assuming you installed using `git`, issue `git checkout development` from the `tuya-convert` folder

alunarkid, 2019-10-10:
Branch 'development' set up to track remote branch 'development' from 'origin'.
Switched to a new branch 'development'

alunarkid, 2019-10-10:
==> ./scripts/smarthack-wifi.log <==
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
hostapd_free_hapd_data: Interface wlan0 wasn't started
AP closed

FF-Fox, 2019-10-11:
Is your wifi interface in fact wlan0?
Try changing it in the config.txt file if not.

alunarkid, 2019-10-11:
I’m pretty sure it is

alunarkid, 2019-10-11:
I’m using a wifi dongle but I don’t know if it’s being recognised

FF-Fox, 2019-10-11:
If you're only pretty sure, I'm definitely not sure...

First of all do a `sudo ifconfig -a` and see if the dongle is recognised.
If it's there, do a `sudo iw list` and check the _Supported interface modes:_ block, **AP** should be listed there.

It will only work if your dongle is detected, it supports AP mode and the interface is correct in the tuya-convert config.txt.

alunarkid, 2019-10-11:
i@raspberrypi:~ $ sudo ifconfig -a
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.106  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::fefa:e916:3ec3:fadf  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:39:06:11  txqueuelen 1000  (Ethernet)
        RX packets 59975  bytes 9937979 (9.4 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0        TX packets 431  bytes 60816 (59.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 18  bytes 1102 (1.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0        TX packets 18  bytes 1102 (1.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500        ether 80:3f:5d:11:45:b2  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

dmascord, 2019-10-15:
What dongle is this?  Sounds like an mt76 or rt8188 based USB dongle, where the stock drivers don't support AP mode.

What does 'iw phy0 info' show ?

        Supported interface modes:
                 * IBSS
                 * managed
                 * AP
                 * P2P-client
                 * P2P-GO
                 * P2P-device

You might need to compile a new wifi module to support your USB dongle to run as an AP, or maybe a new ucode will solve it.

tonysprenk, 2019-10-15:
I think I'm having the same issue. I'm using a TP-Link TL-WN725N on a Kali Linux VM. The weird thing is that I am able to create an accespoint in the WIFI settings but when I run start_flash.sh it says that my wifi dongle doesn't support AP. Any suggestions?

rikthefrog, 2019-10-15:
I have the same with my Raspberry PI 4, strange thing it worked once. After the reboot it didn't worked anymore.

kueblc, 2019-10-29:
Closing since author did not respond, anyone still having issues please follow the process for opening a new issue including your logs

closed 2019-10-29 by kueblc