ct-Open-Source/tuya-convert #310: Teckin sp22 tuyaconvert problem .................................

roberdlv, 2019-10-10:
Hello friends, first of all congratulate on this great community.

Two days ago I bought 4 Teckin SP22 plugs on Amazon, I am trying to convert you https without success, I put the pairing mode (blue fast blinking) and I only get on the screen ............. ..........................

Say I have looked in the smart life app and version appears
Device: Teckin SP22
WIFI Firmware v1.0.1
MCU Firmware v1.0.1

Does this mean that I have a device from You or Realtek? Any way to know unopened? Any alternative to solve my problem? Or do I have to return to Amazon / Teckin? Thanks in advance, I put some images so you can see my screen as it is. Sorry for my English since I am from Spain and I use some translator.
[image]
[image]
[image]

Also say that when I press enter on the command line and teckin sp22 is flashing blue, it immediately turns off after entering and remains off.

If I do the same process but blinking purple / red, immediately after pressing enter on the command line then relay sounds and teckin sp22 goes off.

Thanks for your time

kueblc, 2019-10-10:
Unfortunately this device is using a Realtek chip instead of ESP82xx, so you will not be able to run custom firmware, whether by `tuya-convert` or direct wire.

Also your `tuya-convert` is out of date, so if you have other devices to flash I recommend updating first.

closed 2019-10-10 by kueblc

roberdlv, 2019-10-11:
oh, thanks ... how can you tell what realtek has? and how can i update yours convert? thank you!

Finally there is the same problem with the sp21? or these would be correct to buy and be able to make yours convert and not realtek. Thanks again @kueblc

kueblc, 2019-10-11:
You can tell because it calls `/d.json` not `/gw.json` in your web log.

I do not know what the status of the SP21 is.

kueblc, 2019-10-11:
To update, assuming you installed with `git`, do `git pull`.

If you did not install with `git`, you should.
`git clone https://github.com/ct-Open-Source/tuya-convert.git`

roberdlv, 2019-10-11:
Thanks friend, I think I'm going to ask for Blitzwolf shp2 or shp8.  what do you think?

roberdlv, 2019-10-11:
I tried to flash yours convert a hyleton 314 switch, I started putting ......... online commands and LED indicator plug is off, now the switch does not work, LED stays off and a Wi-Fi appears that puts vtrus recovery, which  have i done wrong?  How do I recover?

@kueblc

kueblc, 2019-10-11:
Check "Troubleshooting" wiki page

kueblc, 2019-10-11:
You need to configure the device, you can seek help for that from Sonoff-Tasmota

roberdlv, 2019-10-11:
Sorry i can't understand, I put Wi-Fi configuration and Wi-Fi pass and now I am not able to enter the plug configuration again, the LED and button do not work and I cannot put ap mode.

@kueblc

kueblc, 2019-10-11:
LEDs and button will do nothing until you configure it.

Check your router to see if the device has connected. Then you'll need to go to that IP address to finish configuration. I can't provide support for that, you'll need to seek help from Tasmota.

If it hasn't, you may have entered the wrong password. There is support for that in the wiki.

roberdlv, 2019-10-11:
ok, this is solved, it was the router assigned mac with random ip.  I'm missing the part of running led and button I have only seen hyleton 313 I have set the same configuration but those gpios do not work

@kueblc thanks!!

roberdlv, 2019-10-12:
The only problem I have now with the hyleton 314 after moving to Tasmota is that it does not work LED or button on manual off ... I have put the GPOI that puts on the page:

https://github.com/ct-Open-Source/tuya-convert/wiki/Compatible-devices-(HTTP-firmware)

Hyleton 314 | EU | wifi smart plug | HLT-314 | GPIO0 = LED2i, GPIO2 = LED1i,GPIO13 = Button1,GPIO15 = Relay1
-- | -- | -- | -- | --

But it doesn't solve the problem, I don't know what to do, if with ESPhome it would be solved or is there any other configuration? Thanks in advance.

thanks for all @kueblc

roberdlv, 2019-10-13:
Is it possible to get correct GPOIS for HYLETON 314 EU? the ones on the page seem to be incorrect nothing works when you put them in and it is automatically restored to factory values ​​after ... Thanks

kueblc, 2019-10-13:
@roberdlv yes it is possible to find the correct GPIO, but you need to seek support from Tasmota. I am a volunteer maintaining `tuya-convert` on my own time, I can not help with unrelated matters.

roberdlv, 2019-10-13:
Thanks friend, I already managed to fix it GPIO was wrong.

I put the configuration that I managed to make work for if someone has HYLETON 314 can also solve their problem.

Thanks for everything.
[image]