ct-Open-Source/tuya-convert #314: Simple connection to smartphone was not enough

Cyber1000, 2019-10-12:
Hi,

Not a bug so far, but a comment on the line 
"Connect any other device (a smartphone or something) to the WIFI $AP"

For me a simple connection from smartphone to vtrust-flash-AP wasn't enough, the device just couldn't be found at 10.42.42.42. I had to generate network traffic by googling on my smartphone. There were no network results, since vtrust-flash-AP doesn't connect to internet, but it seemed to generate enough traffic that tuya-device got 10.42.42.42 und started to backup. 
Had this effect every time.

Just wanted to know, if others experience this too, if yes it would be fine to add an additional comment to start_flash.sh

Thanks!

kueblc, 2019-10-12:
Interesting findings, thank you for sharing. I have not heard of this from others. Will require further investigation as to why you had this experience.

kueblc, 2019-10-12:
Also note that the device will not connect to `10.42.42.42` until it is running our firmware already. When it first connects, it is expected to take an address in the range `10.42.42.10,10.42.42.40`.

Cyber1000, 2019-10-12:
Yes I don't think it got the firmware until I started to generate some traffic. I noticed this behaviour with 2 Gosund SP111 and I waited some minutes before I tried to start googling from my smartphone.

I have some Gosund SP111 still to flash, I'll watch this issue in the next days. Are there any logs which may be interesting?

kueblc, 2019-10-12:
Most (all?) smart phones have *some* traffic going on at all times, if nothing else, a connectivity check. So I'm not sure how the added traffic would change anything. I am at a loss for an explanation here. If you have any additional findings or theories please feel free to share.

Cyber1000, 2019-10-13:
I don't know either, perhaps there is just a short connectivity check not at the same time the flash should happen. The blue flashing of the device just stopped (and the device seemed to reboot) after producing traffic.
Maybe it's a honor/huawei "problem", as far as I know it's pretty aggressive trying to save energy.
I'll look in the logs written by start_flash.sh, somewhere within this week.

Btw. should we put these logfiles in it's own subfolder, like development-branch uses it's own folder for backups? Would help for docker-usages and would only be a small adjustment.

kueblc, 2019-10-13:
Currently doing a rewrite of the logging system, but yes I think we could have the logs written to the same directory as the backups, so that each run is kept organized

kueblc, 2019-10-29:
Closing for housekeeping, thanks again for reporting on your experience

closed 2019-10-29 by kueblc

Cyber1000, 2019-11-17:
With the latest version I didn't have any issues with my smartphone. Either the newest version solved the problem somehow or it was just a random bug, when I tried it back in October.

Another question: Would it make sense to make a PR for saving logs in the same directory as the backups? I have a branch in my repo ... But since you're writing "Currently doing a rewrite of the logging system" it may not make much sense anymore.

kueblc, 2019-11-18:
> With the latest version I didn't have any issues with my smartphone. Either the newest version solved the problem somehow or it was just a random bug, when I tried it back in October.

Awesome!

> Another question: Would it make sense to make a PR for saving logs in the same directory as the backups? I have a branch in my repo ... But since you're writing "Currently doing a rewrite of the logging system" it may not make much sense anymore.

Sounds like a nice feature, I suppose the (minor) challenge is that a new backup directory is created for each flash attempt. It might be tricky to keep all the details pertaining to one device in one place. Perhaps copy and truncate is the best approach?

Cyber1000, 2019-11-19:
My approach was to move the logfiles at the end of the loop into the backupdir, as far as I could see the logfiles are getting recreated. It's not 100% exact, but should get most parts of the logs on a "per device" base. You can see it in https://github.com/Cyber1000/tuya-convert/blob/master/start_flash.sh. Tests are highly welcome, I had 3 devices for this, where it worked. Now they are flashed and I have no more devices for the near future.

kueblc, 2019-11-19:
Seems like a decent approach. If we move them at the very end of the loop, after the flashing menu, I do think it should capture everything.

I would maybe use the glob `*.log` instead of naming specific log files. We added another service that generates a log in the next release (see `development`).

Also since we start `screen` with `sudo`, I'm not entirely sure but the log files might not be owned by the current user, might want to use `sudo mv` too.

Feel free to make a PR against `development` and we can try to get this in before the next release.

Cyber1000, 2019-11-19:
Agree: added line to the end of loop with *.log, sudo is a good idea too (since I'm using a docker-image it didn't make a difference to me)
https://github.com/ct-Open-Source/tuya-convert/pull/411