ct-Open-Source/tuya-convert #316: Usage in docker messes with host network

Cyber1000, 2019-10-12:
Since I didn't want to mess with my host system (and also had no empty raspi, I could use), I made a docker-container. 
This works fine (I may PR it here, if someone is interested), but I had to set privileged and network_mode:host to get this working. 

It seems to change routes on the host, since privileged and network_mode:host gives a lot of control to the container. After stopping the container I have some network-related issues (connection problems to other docker-containers on the same computer). 
Only restart of host helps.

@kueblc - I've seen you're commenting about your branch create-ap to use in ubuntu/docker. What's the difference to this master? I've only seen it has 42 additional commits, but not quite sure what's the purpose of this branch, since this master also works for me on a ubuntu docker/ubuntu host combination besides this side effects.

kueblc, 2019-10-12:
`create_ap` is a third party script maintained by @oblique that makes it dead simple to launch an AP. In the `create-ap` branch I replaced the `setup_ap` script with a call to `create_ap`.

Older versions of `tuya-convert` were a little destructive when setting up an AP, and as I was having trouble getting it to work on my Ubuntu desktop I considered that outsourcing this task would make things easier.

Since #225 I've been able to correct most if not all of the issues, so I've ceased maintaining the `create-ap` branch. I would recommend trying the `development` branch if you haven't already.

Cyber1000, 2019-10-13:
Ok, I'll try the development branch this week, maybe I'll find the problem or perhaps it get's solved by this branch. I'll report then and close this issue.

Thanks for clarification.

kueblc, 2019-10-29:
Closing for now @Cyber1000 but feel free to reopen if you need to follow up

closed 2019-10-29 by kueblc

Cyber1000, 2019-11-17:
I've tried the newest branch and it seems to work, no destructive settings anymore! Just wanted to mention this, sorry for the late reply.

kueblc, 2019-11-17:
Great, thanks for following up!

moritzj29, 2019-11-19:
Hello @Cyber1000,
do you mind sharing your approach using docker? I prefer the idea of docker over using a RPi every time I get some new devices :)
Thanks!

Cyber1000, 2019-11-19:
@moritzj29 No problem, it's a setup separated into Dockerfile (for building a docker-image locally) and docker-compose.yml (for running the container). You have to use it on linux-host (I used ubuntu, but shouldn't matter really if you take a well known linux distro).
Hope the readmes are clear (I've just written them, these repos were for personal use until now). Feel free to ask, if not.

Building docker-image: https://github.com/Cyber1000/Docker-Images-Tuya
Using the container with docker-compose: https://github.com/Cyber1000/Docker-Services-Tuya

Perhaps I'll make a PR somewhere in the future to include the Dockerfile in this project, but there are some minor (mostly cosmetic) points I would change.

Daviteusz, 2020-01-05:
@Cyber1000 thanks for your work. I little edited your work for rpi4 
https://github.com/Daviteusz/Docker-Images-Tuya

And working very good.

Cyber1000, 2020-01-05:
@Daviteusz  You're welcome :-)

Some points:
- You have forked from the master of my repo, which is not up-to-date (and will be uptodate only now and then) and was a first approach, you may fork from the development branch (and somewhere in the future from the master branch) of this repo: https://github.com/ct-Open-Source/tuya-convert/tree/development
- If you do a RUN apt-get update and separately do a RUN apt-get install ... the RUN apt-get update will only be run once on your computer (as a cached layer by docker) and so you won't get fresh installs on subsequent docker-builds. See more at https://runnable.com/blog/9-common-dockerfile-mistakes (point 1)
- why I choose phusion/baseimage (actually a adjusted ubuntu 16.04): ubuntu isn't made for living inside docker and some things like init-process don't work fully reliable, see https://hub.docker.com/r/phusion/baseimage. But this is more a problem of docker-containers which are started and run for days and weeks and not a problem with this short-executing-scripts, so you should be fine.

**I assume the phusion/baseimage:0.11 part was the tricky part why it didn't work for rpi4?**