ct-Open-Source/tuya-convert #317: Moes WiFi Thermostat not flashing

31warren, 2019-10-13:
I have successfully flashed several power monitoring sockets but a 'Moes' branded WiFi thermostat (like this one https://github.com/klausahrenberg/ThermostatBecaWifi ) isn't flashing.

The SmartConfig log file shows SmartConfig complete.

The Web log shows:

^CTraceback (most recent call last):
  File "./fake-registration-server.py", line 129, in <module>
    main()
  File "./fake-registration-server.py", line 125, in main
    tornado.ioloop.IOLoop.current().start()
  File "/usr/lib/python3/dist-packages/tornado/platform/asyncio.py", line 132, in start
    self.asyncio_loop.run_forever()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 539, in run_forever
    self._run_once()
  File "/usr/lib/python3.7/asyncio/base_events.py", line 1739, in _run_once
    event_list = self._selector.select(timeout)
  File "/usr/lib/python3.7/selectors.py", line 468, in select
    fd_event_list = self._selector.poll(timeout, max_ev)
KeyboardInterrupt
Listening on port 80
^[[32m[I 191013 12:55:02 web:2162]^[[m^O 200 GET / (10.42.42.23) 4.48ms

The flash program doesn't stop, just keeps adding more ... to the screen.

I have tried the curl flash3 command using another terminal window but 10.42.42.42 isn't open

Any suggestions please?

kueblc, 2019-10-13:
Not enough information, please read and follow the directions when opening a new issue. Please check the wiki for troubleshooting tips and if you still can't figure it out, post your logs.

dmascord, 2019-10-15:
While the dots are showing, can you confirm that you have a route for 10.42.42.0/24 associated with wlan0?

ie, ip route should show a line like:
10.42.42.0/24 dev wlan0 scope link

If that route doesn't exist, then you will need to create it, ie ip route add 10.42.42.0/24 dev wlan0

31warren, 2019-10-17:
Thanks for the help.

I have now flashed the chip manually instead of OTA.

I have ordered another thermostat and will try again when the new one arrives.

kueblc, 2019-10-18:
Glad you figured something out. Please remember to close your issue.

closed 2019-10-18 by kueblc