ct-Open-Source/tuya-convert #318: could not establish sslpsk socket

madprunes, 2019-10-15:
Getting the following error in the smarthack-psk.log file, the script just sits there printing . endlessly, both the Mirabella Genio and Brilliant Smart bulbs connect to the WIFI but that's as far as they get.

```
new client on port 443 from 10.42.42.23:2807
('could not establish sslpsk socket:', AttributeError("'module' object has no attribute 'wrap_socket'",))
new client on port 443 from 10.42.42.23:37832
('could not establish sslpsk socket:', AttributeError("'module' object has no attribute 'wrap_socket'",))
```

dmascord, 2019-10-15:
What version of Python are you using?  This looks like a pre-requisite missing issue.

madprunes, 2019-10-15:
The installed versions are
```
pi@raspberrypi:~ $ python --version
Python 2.7.13
pi@raspberrypi:~ $ python3 --version
Python 3.5.3
```

madprunes, 2019-10-15:
I just noticed M4dmartig4n's psk-frontend.py specifies python3 but the this one doesn't , I can try changing it when I'm not at work and have access to a light socket.

madprunes, 2019-10-17:
Yes I found the same thing when changing it to python3 I got that error.
I copied M4dmartig4n's psk-frontend.py into the directory and used that instead and the lights got to the point of MQTT and Web calls, but I don't think either of the lights I have are supported yet.

sanderdekroon, 2019-10-18:
That's weird, have all changes from M4dmartig4n not been merged? Because my device would not even connect.

I'll try his python file tonight and see if that resolves anything.

sam7323, 2019-10-29:
I got similar error when flashing Zemismart 2.5" wifi downlight - PSK: ec2ee8a...... ('could not establish sslpsk socket:', error(111, 'Connection refused')) new client on port 443 from 10.42.42.32:47423.

On further inspection found that the start_flash.sh has hardcoded ip to listen and burn flash at 10.42.42.42 whereas in setup_ap.sh under scripts, DHCP range is 10.42.42.10 to 10.42.42.40 with 12h lease time. The smart light was assigned an ip of 10.42.42.32 and hence refusing secure connection.

Once DHCP range changed to 10.42.42.50, start_flash.sh found Zemismart smart light at 10.42.42.42 and Tasmato was burned successfully.

kueblc, 2019-10-29:
> I got similar error when flashing Zemismart 2.5" wifi downlight - PSK: ec2ee8a...... ('could not establish sslpsk socket:', error(111, 'Connection refused')) new client on port 443 from 10.42.42.32:47423.

This is a different issue. Are you sure the device was at .32 and not your phone?

> On further inspection found that the start_flash.sh has hardcoded ip to listen and burn flash at 10.42.42.42 whereas in setup_ap.sh under scripts, DHCP range is 10.42.42.10 to 10.42.42.40 with 12h lease time.

This is intentional so that the device won't accidentally get the 10.42.42.42 address which has a special meaning here. It's used only by the intermediate firmware and not by the stock firmware, so that we know when the device is running the intermediate firmware.

> The smart light was assigned an ip of 10.42.42.32 and hence refusing secure connection.

This just means the device didn't install the intermediate firmware yet, still running Tuya

> Once DHCP range changed to 10.42.42.50, start_flash.sh found Zemismart smart light at 10.42.42.42 and Tasmato was burned successfully.

Not a safe change to make for the aforementioned reasons, but glad you were able to proceed with your flashing.

sam7323, 2019-10-29:
> > I got similar error when flashing Zemismart 2.5" wifi downlight - PSK: ec2ee8a...... ('could not establish sslpsk socket:', error(111, 'Connection refused')) new client on port 443 from 10.42.42.32:47423.
> 
> This is a different issue. Are you sure the device was at .32 and not your phone?
The phone was at .29
> This is intentional so that the device won't accidentally get the 10.42.42.42 address which has a special meaning here. It's used only by the intermediate firmware and not by the stock firmware, so [...]

Who assigns 10.42.42.42 to smart light? VTRUST dhcp running on Pi or the script?
> 
> > The smart light was assigned an ip of 10.42.42.32 and hence refusing secure connection.
> 
> This just means the device didn't install the intermediate firmware yet, still running Tuya
> 
OK, I got above error since last night on several clean Buster Lite and Stretch Lite installs on Pi 3 B+ model.

> > Once DHCP range changed to 10.42.42.50, start_flash.sh found Zemismart smart light at 10.42.42.42 and Tasmato was burned successfully.
> 
> Not a safe change to make for the aforementioned reasons, but glad you were able to proceed with your flashing.

As I mentioned above, once changed to 10.42.42.50, the smart device found immediately and flashing was successful! As a matter of fact, I flashed 10 in total!

Still not sure how 10.42.42.42 is assigned every single time to smart device? Is it static assigned or by DHCP server run by VTRUST?

kueblc, 2019-10-29:
The intermediate firmware (`upgrade.bin`) requests 10.42.42.42 once installed by `tuya-convert`

kueblc, 2019-11-03:
@madprunes were you able to resolve your issue?

madprunes, 2019-11-03:
I haven't had a lot of time to work through it, my immediate need was rectified by opening the bulbs and hardware flashing them, I did pull the software as even with the hobbled together version of tyua-convert which the lights connected to the firmware never seemed to get pushed.

kueblc, 2019-11-03:
Would you mind sharing that backup for testing and debugging?

madprunes, 2019-11-03:
Yes, I've been meaning to upload them, I should get an opportunity in the next 12 or so hours.

madprunes, 2019-11-04:
Here are the firmwares I have pulled.

[mirabellagenio-rgbw-b22.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/3803343/mirabellagenio-rgbw-b22.bin.gz)
[mirabellagenio-rgbw-e27.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/3803344/mirabellagenio-rgbw-e27.bin.gz)

kueblc, 2019-11-07:
Thanks for uploading @madprunes. Unfortunately I wasn't able to reproduce the issue here. Would you share the rest of your log files to see if that has any more clues?

madprunes, 2019-11-07:
@kueblc Out of interest what OS are you using and was it just that latest tuya-convert?, I will endeavour to set it up and try again this weekend with another bulb which hasn't been hardware flashed, and I'll report back.

kueblc, 2019-11-07:
Ubuntu 18.04 x64 on my development machine, with the latest `tuya-convert`

barrymossel, 2019-11-13:
Getting the same error:

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3842979/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3842980/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3842981/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3842982/smarthack-psk.log)

Was this ever solved? https://github.com/ct-Open-Source/tuya-convert/issues/390

kueblc, 2019-11-13:
This is not the same error @barrymossel, @madprunes was getting `'module' object has no attribute 'wrap_socket'` while you have `NO_SHARED_CIPHER`. This is coming from your phone, not the device, and has no impact on the process. You can safely ignore it.

barrymossel, 2019-11-13:
Sorry, will continue in other issue!

kueblc, 2019-11-17:
@madprunes did you have any problems installing `sslpsk`? Can you test that it has a `wrap_socket` method in `ipython2`?

kueblc, 2019-11-21:
Closing due to inactivity, but please reopen if there is more information to share

closed 2019-11-21 by kueblc

MikeJMcGuire, 2024-07-05:
So the current version of tuya-convert, sslpsk (from drbild), and python3 give the original error at the start of this issue - sslpsk doesn't have a wrap_socket method (called by psk-frontend.py). This can be worked around by using a fixed version of sslpsk that adds a wrap_socket method back in. @doronz88 added a fix which hasn't yet been applied to @drbild original repository. I've forked it to https://github.com/MikeJMcGuire/sslpsk and then modified the install_prereqs.sh file to refence my repository instead of the original. Then I was able to get it to work on a recent install of Ubuntu with python3.