ct-Open-Source/tuya-convert #320: Unable to flash LSC SmartConnect LED Strip

Pannnenkoek, 2019-10-16:
Hey everyone,

Last week I've bought a LSC SmartConnect led strip at Action (LSC, 970750), which uses the Tuya platform. I want to flash it so it can be used with HomeAssistant via MQTT instead of relying on the native Chinese platform ofcourse.

The device can be placed into pairing/flashing mode by quicky plugging and unplugging the powercable three times in succession, afterwards the led strip will flash rapidly.

The AP vtrust-flash is online and I can connect to it with a smartphone, and the device is in flashing mode. It looks like SmartConfig is succesfully flashed but the proces is stuck on "Waiting for the upgraded device to appear":

`======================================================
Starting pairing procedure in screen
RTNETLINK answers: File exists
~/tuya-convert
Waiting for the upgraded device to appear
If this does not work have a look at the '*.log'-files in the 'scripts' subfolder!
`
I've already tried to use another branch which solved the flashing problem for another user (can't come up with the name anymore) and I've double checked the compatibility of the device. Can anyone care to take a look at the logs and help me out? Many thanks in advance.

**Logfiles**
[smarthack-mqtt.log](https://pastebin.com/WFFGMj9b)
[smarthack-web.log](https://pastebin.com/x3VQ2Y9x)
[smarthack-smartconfig.log](https://pastebin.com/eWTx6A2Y)
[smarthack-psk.log](https://pastebin.com/iYaVhiTy)
[smarthack-wifi.log](https://pastebin.com/eiTvyk2F)

kueblc, 2019-10-16:
According to your web log, you have another process running on port 80.

Pannnenkoek, 2019-10-21:
> 
> 
> According to your web log, you have another process running on port 80.

I missed that. I've killed the proces responsible for hogging port 80 and tried again, but to no avail. I've also added ports 80,42,1883 and 8883 to my firewall since I just remembed I have ufw installed but after trying for a while, I still can't get the led strip to connect.

Any other ideas? Seems more people are having [this issue.](https://github.com/ct-Open-Source/tuya-convert/issues/323)

kueblc, 2019-10-22:
@Pannnenkoek @DerpyDoom @Vibestar

Can you retrieve the `productKey` or `schema` for the device? You could do this by capturing the device registering with the cloud, using `tuyapi`, or through a firmware backup.

Vibestar, 2019-10-22:
I’m going to return the led strip. It has a huge powerdrop in the strip. First led is bright and the last one half of the brightness. That’s no good. 
Hopefully you’ll find a answer.

DoomStyler, 2019-10-22:
Well it's a cheap Brand, expect something like this.

I'll try to get the productkey or schena for You. Will Update If I succeed.

Vibestar, 2019-10-22:
Yeah, but this dimming and the fact that it can’t pair yet is 20 euro’s to much.

Speedbreaker12, 2019-10-27:
I was able to flash 4 of those devices successfully now, so it's definitely working.

DerpyDoom, 2019-10-27:
Feedback: I managed to flash them via a vmware linux vm, it looks like this doesn't work via pi zero. I managed to brick mine by flashing a new version of "sonoff.bin", any idea how i can retrieve it?

tonysprenk, 2019-10-27:
@DerpyDoom I’ve been trying to get it working on a MacBook Pro with a Linux VM but no luck. What’s your setup?

DerpyDoom, 2019-10-27:
> @DerpyDoom I’ve been trying to get it working on a MacBook Pro with a Linux VM but no luck. What’s your setup?

I'm running Windows 10 on my main machine, debian 10 as vm host.
For the WiFi hotspot I'm using this one [here](https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/de/wireless_adapters_n150/ew-7811un/) that is being detected as soon as I plug it in. Vmware asks me If I want to mount it to the VM and of course I click yes.

Don't forget to edit your wifi card in the config of tuya-convert (can be retrieved by ip -a). And I also have noticed if You are trying things out multiple times it is possible that not every process from tuya-convert is being stopped. So you might need to kill them by hand.

kueblc, 2019-11-03:
@Pannnenkoek is this resolved?

closed 2019-11-10 by kueblc

sehraf, 2019-12-15:
Today I've successfully flashed my led strip but encountered some weird problems i like to share for the sake of record.
My attempt to take over this tuya device failed due to an **unanswered TCP handshake**!

But first thinks first: 
* Everything was setup and running
* The LED strip received the WLAN credentials
* It joined the vtrust-flash WLAN
* It asked for the IP of the tuya cloud using DNS
* It **tried** to connect to the (fake) tuya cloud over https

And this is were it got fishy. The logs didn't show anything so i used wireshark and saw the communication: The ESP was sending a TCP SYN which was never answered by the psk-thingy (multiple times). When using `nmap` of `curl` i could connect to the psk-thingy (and when using http (no s!) i even got an error from the psk script (due to TLS handshake failure which was expected)). So the python side seems to be fine. 
I have no idea what caused this. I "fixed" it by booting Kali Linux from a stick (i'm on Arch).

So, in case anybody experiences this: try to use Kali Linux!
Indications for are:
* The device leaves pairing mode (stops blicking)
* The wifi process detects a new device joining
* Nothing more, all logs are quiet