ct-Open-Source/tuya-convert #322: Teckin SB50 just won't connect

tatsumara, 2019-10-18:
I just can't get the SB50 to connect, at all.
I've been trying for hours now, sifting through every mention of this script i could find, with no success.
Hotspot is working, although for some reason the vtrust-flash network has a connection to the internet.

I'm using the script on an RPi 3B+ running DietPi.

Name: LED Smart Bulb
Device: Teckin SB50
Version: 1.0.5

[smarthack-smartconfig.log](https://github.com/ct-Open-Source/tuya-convert/files/3745698/smarthack-smartconfig.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/3745699/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/3745700/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/3745701/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/3745702/smarthack-psk.log)

kueblc, 2019-10-18:
Have you tried the `development` branch?

tatsumara, 2019-10-18:
Development branch shows a lot more activity, but ends with"Device did not appear with the intermediate firmware".

tatsumara, 2019-10-18:
Scratch all that, for some reason it has already flashed the Sonoff firmware and is working like expected!

closed 2019-10-18 by tatsumara

kueblc, 2019-10-18:
Happy to hear it!