ct-Open-Source/tuya-convert #323: LSC SmartConnect LED Strip not flashable

DerpyDoom, 2019-10-20:
I am trying to flash a LSC SmartConnect LED Strip from a store called Action in Germany.

I don't seem to succeed with it. Using a Raspberry Pi Zero W via Usb Ethernet. I can connect my phone, set my strip to pairing mode and press enter. The Light also stops blinking but nothing except "Waiting for device to appear........"

Logs:
[smarthack-mqtt.log](https://pastebin.com/x7UUVTQS)
[smarthack-psk.log](https://pastebin.com/Xc47z9cY)
[smarthack-smartconfig.log](https://pastebin.com/DySRVB75)
[smarthack-web.log](https://pastebin.com/TwwzCmHz)
[smarthack-wifi.log](https://pastebin.com/bESXgV0J)

Vibestar, 2019-10-21:
I have the same problemm. The chip in the LSC LED strip seems to be a TYWE3L. That seems to have an ESP8266 on board. Strange that it won't connect.

kueblc, 2019-10-22:
Duplicate of #320, let's keep the conversation going there

closed 2019-10-22 by kueblc