ct-Open-Source/tuya-convert #326: install_prereq.sh fails when installing sslpsk

bdcarr, 2019-10-21:
Hello mate, running Ubuntu in the WSL. As shown below, the prereq script fails. The directory containing `git` is in my $PATH.

```sh
cd ~/Downloads
git clone https://github.com/ct-Open-Source/tuya-convert
cd tuya-convert
./install_prereq.sh
```
Output:
```Hit:1 http://archive.ubuntu.com/ubuntu bionic InRelease
Get:2 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
Get:3 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
Get:4 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [74.6 kB]
Fetched 252 kB in 3s (77.4 kB/s)
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  git-man libcurl3-gnutls liberror-perl libfreetype6
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  ubuntu-wsl wslu
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
net-tools is already the newest version (1.60+git20161116.90da8a0-1ubuntu1).
python-dev is already the newest version (2.7.15~rc1-1).
python-setuptools is already the newest version (39.0.1-2).
python3-setuptools is already the newest version (39.0.1-2).
dnsmasq is already the newest version (2.79-1).
haveged is already the newest version (1.9.1-6).
python3-wheel is already the newest version (0.30.0-0.2).
curl is already the newest version (7.58.0-2ubuntu3.8).
libssl-dev is already the newest version (1.1.1-1ubuntu2.1~18.04.4).
python3-dev is already the newest version (3.6.7-1~18.04).
screen is already the newest version (4.6.2-1ubuntu1).
hostapd is already the newest version (2:2.6-15ubuntu2.5).
mosquitto is already the newest version (1.4.15-2ubuntu0.18.04.3).
python-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.1).
python3-pip is already the newest version (9.0.1-2.3~ubuntu1.18.04.1).
The following packages were automatically installed and are no longer required:
  git-man libcurl3-gnutls liberror-perl libfreetype6
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Collecting git+https://github.com/M4dmartig4n/sslpsk.git
  Cloning https://github.com/M4dmartig4n/sslpsk.git to /tmp/pip-bjsh9pjd-build
  Error [Errno 2] No such file or directory: 'git': 'git' while executing command git clone -q https://github.com/M4dmartig4n/sslpsk.git /tmp/pip-bjsh9pjd-build
Cannot find command 'git'```

bdcarr, 2019-10-21:
This might be relevant:
[image]

frostworx, 2019-10-22:
looks like you need to install "git"

kueblc, 2019-10-22:
I'm not familiar with WSL and it's not one of our supported environments. If you do figure it out, please share so others can benefit in the future.

bdcarr, 2019-10-23:
Update on this, I've narrowed it down a bit. The issue is that when I run a command with `sudo`, it can't locate `git`.

[image]

I have a feeling it's because I installed git via linuxbrew, because I'm a wanker. I'll try removing it and installing via `dnf` instead.

closed 2019-10-23 by bdcarr

kueblc, 2019-10-23:
Good to know, thanks for following up @bdcarr!

bdcarr, 2019-10-27:
Thought I should give another update, in case anyone else was trying the same thing as me.

It doesn't look like this will work in WSL yet, because WSL kind of abstracts the network interfaces a bit. Instead of showing up as `eth0, eth1, wlan0`, they're all ethernet, i.e. `eth0, eth1, eth2`.

I didn't poke too hard at this though due to time constraints; maybe it can be worked around by renaming the correct one to `wlan0`.