ct-Open-Source/tuya-convert #328: Thirdparty bin build by Arduino having issues.

crazycj, 2019-10-22:
Hi all,

I'm having issues uploaden my own thirdparty.bin using the tuya converter. The upload is done successful but there is no response coming from the light that I flash. If I flash it after using a serial USB with TX and RX connection then the same bin uploads and works directly.
Also when I install the tasmota.bin using the tuya converter way then the light also functions.

Are there any requirements on how to build your own BIN? I'm currently using Arduino IDE to build it.

Hope you guys can give me some help/advice.

Thanks!

kueblc, 2019-10-22:
Could you share more information please?

What command are you using to upload the bin? Can you share the bin/source with us? Do you have a backup of the original firmware you can post?

Jason2866, 2019-10-24:
@crazycj  Every 3rd party firmare can fail after Tuya Convert has loaded. 
Reason: The not occupied flash from the 3rd party firmware is not erased from Tuya convert.
In this flash area are the fragments of the orig. firmware. This fragments can make problems
because some parts in flash are used from 3rd party firmwares to store there settings.

Jason2866, 2019-10-24:
Device has problems after Tuya Convert is run and Tasmota is loaded without any precautions done.
[image]
then a TON of FFFFFFF's
[image]
Luckily device starts!!! After a `reset 5` Tasmota command (erases flash behind Tasmota)
**cleans it up and no more errors!!**
[image]

So every device that starts when  converted with Tuya Convert is a lucky punch!

kueblc, 2019-10-24:
@Jason2866 they clearly state that they had no problem with the tasmota bin

> Also when I install the tasmota.bin using the tuya converter way then the light also functions.

This is about a custom program they've written

> I'm having issues uploaden my own thirdparty.bin using the tuya converter.
> Are there any requirements on how to build your own BIN? I'm currently using Arduino IDE to build it.

crazycj, 2019-10-25:
Hi all,

Thanks for your replies.
@kueblc I changed the name from thirdparty.bin to tasmota.bin and then uploaded my own thirdparty,bin to the raspberry pi. When connected to light bulb I run the following command:
curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/thirdparty.bin
Then the light crashes and doesn't boot up my own bin.

If I run the command:
curl http://10.42.42.42/flash3?url=http://10.42.42.1/files/tasmota.bin then everything works fine and I can connect to the light by wifi.

For my own BIN I use DIYHUE, compiled bin is here:
[hue_rgbw.ino.generic.zip](https://github.com/ct-Open-Source/tuya-convert/files/3771912/hue_rgbw.ino.generic.zip) Had to add it in a zip else github doesn't support it.

@Jason2866 that is your tasmota console right? I'm coming there when I upload the tasmota.BIN but not when I upload my own.

Jason2866, 2019-10-25:
@crazycj  Serial console when Tasmota is booting. The FFFFFFF issue is fixed with latest commit.
So if your firmware doesnt load / boot you have overlooked something

crazycj, 2019-10-29:
@Jason2866 

I'm using the following Arduino code which I compile to bin. See any issues with that?

`/*
  This can control bulbs with 4 pwm channels (red, gree, blue, warm white and could wihite). Is tested with MiLight colors bulb.
*/
#include <FS.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ESP8266HTTPUpdateServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ArduinoJson.h>

// Define your white led color temp here (Range 2000-6536K).
// For warm-white led try 2000K, for cold-white try 6000K
#define WHITE_TEMP 4500 // kelvin

IPAddress address ( 192,  168,   0,  95); // choose an unique IP Adress
IPAddress gateway ( 192,  168,   0,   1); // Router IP
IPAddress submask (255, 255, 255,   0);

#define light_version 2.01
#define PWM_CHANNELS 4

struct state {
  uint8_t colors[PWM_CHANNELS], bri = 100, sat = 254, colorMode = 2;
  bool lightState;
  int ct = 200, hue;
  float stepLevel[PWM_CHANNELS], currentColors[PWM_CHANNELS], x, y;
};

//core

#define entertainmentTimeout 1500 // millis

state light;
bool inTransition, entertainmentRun, useDhcp = true;
byte mac[6], packetBuffer[8];
unsigned long lastEPMillis;

//settings
char *lightName = "New Hue RGBW light";
uint8_t scene = 0, startup = false, onPin = 1, offPin = 3, pins[] = {12, 13, 14, 4}; //red, green, blue, could white, warm white
bool hwSwitch = false;

ESP8266WebServer server(80);
WiFiUDP Udp;
ESP8266HTTPUpdateServer httpUpdateServer;

void convert_hue()
{
  double      hh, p, q, t, ff, s, v;
  long        i;

  light.colors[3] = 0;
  s = light.sat / 255.0;
  v = light.bri / 255.0;

  if (s <= 0.0) {      // < is bogus, just shuts up warnings
    light.colors[0] = v;
    light.colors[1] = v;
    light.colors[2] = v;
    return;
  }
  hh = light.hue;
  if (hh >= 65535.0) hh = 0.0;
  hh /= 11850, 0;
[... 638 lines cut ...]
    light.currentColors[3] = 0;
    if (!entertainmentRun) {
      entertainmentRun = true;
    }
    lastEPMillis = millis();
    Udp.read(packetBuffer, packetSize);
    for (uint8_t color = 0; color < 3; color++) {
      light.currentColors[color] = packetBuffer[color + 1];
      analogWrite(pins[color], (int)(packetBuffer[color + 1] * 4));
    }
  }
}

void loop() {
  server.handleClient();
  if (!entertainmentRun) {
    lightEngine();
  } else {
    if ((millis() - lastEPMillis) >= entertainmentTimeout) {
      entertainmentRun = false;
      processLightdata(4);
    }
  }
  entertainment();
}`

crazycj, 2019-11-13:
No one?

kueblc, 2019-11-16:
I may finally have an answer. You said it worked fine over serial but not OTA. If you used `esptool.py` to do the flashing, it will transparently correct the flash settings, but prior to version 1.5 our flasher did not. You may have compiled your firmware with the wrong flash mode.

kueblc, 2019-11-17:
Hi @crazycj, have you checked to see if the flash mode was the issue? It should be DOUT for maximum compatibility, or whatever settings the `device-info` log gave.

kueblc, 2019-11-21:
Closing due to inactivity, feel free to reopen if this is still unresolved

closed 2019-11-21 by kueblc