ct-Open-Source/tuya-convert #329: No wifi connection available after using Linux laptop for Tuya-convert

GsynthChris, 2019-10-23:
Hello, I tried going through all the closed issues and I don’t see anything similar. 

I am running Linux Mint 19 on my laptop. Unfortunately I used my laptop (that I use!) instead of an rpi to be my vtrust ap

After the tuya-concert process is complete!
When I run ifconfig -a I see my WiFi card but I do not have an option for reconnecting to my WiFi after the process is complete.  I don’t even get any thing if I click rescan for WiFi networks. 

I did run ./stop-flash or whatever but no improvement. 
I’m going to start digging through the scripts to find out what exactly you have done to me! 😂😝😝 (lol jk)

Any help would be appreciated so much!!!!!

kueblc, 2019-10-23:
Your network-manager config was probably overwritten.

You can check if there is a backup at `/etc/NetworkManager/NetworkManager.conf.backup` but that too may have been overwritten if you ran `start_flash` more than once. If so you can open `/etc/NetworkManager/NetworkManager.conf` and remove the `unmanaged` line.

Then run `sudo service network-manager restart` and with any luck you should be back online.

I'd recommend trying the `development` branch out for any flashing you have left to do since this should already be fixed there.

GsynthChris, 2019-10-28:
HA... yeah took a look at the script and saw that it was overwriting my networkmanager.conf because yes I was running start-flash repeatedly. MY fix was to just copy the networkmanager.conf from one of my other PCs cause I didn't know what that is supposed to look like. too bad I didnt check my github before I did that... your way is alot easier. hahaha.  also funny story, I am flashing 4 smart bulbs now and to avoid that happening again i A: copied my networkmanager.conf separately and also I am running stop_flash after each flash. I will fosho check out that dev version though!

kueblc, 2019-10-28:
Glad it worked out, FYI I pushed the latest version so you can just update to get the fix and more

closed 2019-10-28 by kueblc