ct-Open-Source/tuya-convert #332: Quick note on your most recent merge

vishnubob, 2019-10-23:
Hi there,

I see you just pushed a new merge, and I just wanted to let you know I needed to make one adjustment to the sync server to make this go:

```
diff --git a/scripts/fake-registration-server.py b/scripts/fake-registration-server.py
index 2ec144c..45569d4 100755
--- a/scripts/fake-registration-server.py
+++ b/scripts/fake-registration-server.py
@@ -82,7 +82,7 @@ class JSONHandler(tornado.web.RequestHandler):
                 't': ts,
                 'success': True }
             answer = jsonstr(answer)
-            payload = b64encode(AES.new(options.secKey, AES.MODE_ECB).encrypt(pad(answer))).decode()
+            payload = b64encode(AES.new(options.secKey.encode("utf-8"), AES.MODE_ECB).encrypt(pad(answer).encode("utf-8"))).decode()
             signature = "result=%s||t=%d||%s" % (payload, ts, options.secKey)
             signature = hashlib.md5(signature.encode()).hexdigest()[8:24]
             answer = {
``

Otherwise, I got this error:

```
    Traceback (most recent call last):
      File "/usr/local/lib/python3.7/dist-packages/tornado/web.py", line 1697, in _execute
        result = method(*self.path_args, **self.path_kwargs)
      File "./fake-registration-server.py", line 224, in post
        self.reply(answer, encrypted)
      File "./fake-registration-server.py", line 85, in reply
        payload = b64encode(AES.new(options.secKey.encode("utf-8"), AES.MODE_ECB).encrypt(pad(answer))).decode()
      File "/usr/local/lib/python3.7/dist-packages/Crypto/Cipher/_mode_ecb.py", line 135, in encrypt
        c_uint8_ptr(plaintext),
      File "/usr/local/lib/python3.7/dist-packages/Crypto/Util/_raw_api.py", line 235, in c_uint8_ptr
        raise TypeError("Object type %s cannot be passed to C code" % type(data))
    TypeError: Object type <class 'str'> cannot be passed to C code
```

I think there are other places that needed this, but this change was the bare-minimum required in order for me to get the new version working.  Thanks again for this tool, it rules not having to split some of these devices open in order to prop up new firmware.  Great work!

kueblc, 2019-10-24:
Hi @vishnubob thanks for the report. That's odd, that line wasn't touched in the latest release. I'll see if I can reproduce it. What environment are you working with?

vishnubob, 2019-10-24:
@kueblc This is on a clean install of Raspbian Buster after using your provisioning script.  I was literally working on a pair of switches when you pushed your update.  I googled the error, which led me to the recommendation of [byte encoding](https://stackoverflow.com/questions/50302827/object-type-class-str-cannot-be-passed-to-c-code-virtual-environment/50302966) the string:

kueblc, 2019-10-24:
While unnecessary in my environment, it also doesn't seem to hurt. Just pushed this change, thanks for reporting @vishnubob

closed 2019-10-24 by kueblc

frostworx, 2019-11-15:
hi,
thanks for the fix, vishnubob, and thanks for tuya-convert, kueblc!
looks like you haven't pushed the changes to master yet, kueblc:
https://github.com/ct-Open-Source/tuya-convert/blob/master/scripts/fake-registration-server.py
after having installed python2-sslpsk for python2 only psk-frontend.py and applied above patch I successfully flashed 4 gosund sp1 devices on my arch linux machine.
following error still remains, but obviously isn't fatal:

`Traceback (most recent call last):
  File "./mq_pub_15.py", line 98, in <module>
    sys.exit(main())
  File "./mq_pub_15.py", line 93, in main
    m1 = iot_enc(message, localKey, protocol)
  File "./mq_pub_15.py", line 32, in iot_enc
    messge_enc = encrypt(message, local_key)
  File "./mq_pub_15.py", line 24, in <lambda>
    encrypt = lambda msg, key: AES.new(key, AES.MODE_ECB).encrypt(pad(msg))
  File "/usr/lib/python3.8/site-packages/Crypto/Cipher/AES.py", line 232, in new
    return _create_cipher(sys.modules[__name__], key, mode, *args, **kwargs)
  File "/usr/lib/python3.8/site-packages/Crypto/Cipher/__init__.py", line 79, in _create_cipher
    return modes[mode](factory, **kwargs)
  File "/usr/lib/python3.8/site-packages/Crypto/Cipher/_mode_ecb.py", line 215, in _create_ecb_cipher
    cipher_state = factory._create_base_cipher(kwargs)
  File "/usr/lib/python3.8/site-packages/Crypto/Cipher/AES.py", line 103, in _create_base_cipher
    result = start_operation(c_uint8_ptr(key),
  File "/usr/lib/python3.8/site-packages/Crypto/Util/_raw_api.py", line 144, in c_uint8_ptr
    raise TypeError("Object type %s cannot be passed to C code" % type(data))
TypeError: Object type <class 'str'> cannot be passed to C code`

kueblc, 2019-11-15:
Thanks, good catch @frostworx f2499d24f4aed1720d37bbfc90255d5b46c89e96

frostworx, 2019-11-15:
thanks for the quick fix! have a nice weekend @kueblc :)

commit c3952af9 (2019-10-24): Encode AES key

Closes #332