ct-Open-Source/tuya-convert #333: gosund SP111 seems to get SmartConfig, but getting "device did not appear"

bfour, 2019-10-24:
Everything works as expected, the plug is blinking rapidly with a blue light, I start the script, connect a phone and press ENTER. After a few seconds, the plug stops blinking. The script, however, says "Device did not appear with the intermediate firmware". This is the output after pressing ENTER:

> Starting smart config pairing procedure
> Waiting for the device to install the intermediate firmware
> Put device in EZ config mode (blinking fast)
> Sending SSID                  vtrust-flash
> Sending wifiPassword
> Sending token                 00000000
> Sending secret                0101
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> SmartConfig complete.
> Resending SmartConfig Packets
> .................
> SmartConfig complete.
> Resending SmartConfig Packets
> ................
> Device did not appear with the intermediate firmware
> Check the *.log files in the scripts folder
> Stopping smart config
> Do you want to flash another device? [y/N]

Device: gosund SP111 Smart Socket
Firmware: unknown, not visibly specified on packaging or device

[smarthack-mqtt.log](https://pastebin.com/EqiZTtQa)
[smarthack-psk.log](https://pastebin.com/fzS9wHWd)
[smarthack-web.log](https://pastebin.com/4WD6cM1Q)
[smarthack-wifi.log](https://pastebin.com/t2561pEU)

kueblc, 2019-10-24:
According to your web log, you have another process running on port 80. After you close that process and restart `start_flash`, you don't need to repeat the smart config process, just plug the device back in and it should connect.

bfour, 2019-10-24:
I checked but there doesn't seem to be any unless I'm missing something:
[image]

kueblc, 2019-10-25:
Ah you're right, I read too quickly. It's not starting because the address is unavailable, probably because of the web server started before the AP finished setting up. You can try adding a delay [here](https://github.com/vtrust-de/esp8266-ota-flash-convert/blob/master/src/esp8266-ota-flash-convert.ino#L247) or starting the web server manually after the AP is already running.

kueblc, 2019-10-25:
You could also try adding `--addr=0.0.0.0` [to the end of the line here](https://github.com/ct-Open-Source/tuya-convert/blob/3b2b5b8f4ecaca0f88ee2882d3e21fd36822b281/start_flash.sh#L50).

vador98, 2019-10-25:
I'm having the same issue.  I tried adding the 0.0.0.0, but unfortunately receive the same result.

kueblc, 2019-10-25:
Same result as in the same line in your web log?

vador98, 2019-10-25:
Embarassingly, I didn't look at the web log to see the original error here.  No, the error is apparently different, but the items displayed on screen are the same.

kueblc, 2019-10-25:
All issues will have the same "device did not appear" message, you'll need to get the logs out to determine the actual cause

vador98, 2019-10-25:
I'll start another issue and post my logs.  Thanks! :)

bfour, 2019-10-26:
> setting up. You can try adding a delay [here](https://github.com/ct-Open-Source/tuya-convert/blob/3b2b5b8f4ecaca0f88ee2882d3e21fd36822b281/start_flash.sh#L46) or starting the web server manually after the AP is already runni

👍 Quick'n'dirty, but this did the trick. Thanks for you help! 😁

commit 8edc2512 (2019-11-02): Fix #333 #359

Delay start of web server for address to be available
closed 2019-10-26 by bfour